realsenseai/realsense-ros #735: How to remove veiling effect created by the edge of a wall

Patrick-Lascombe, 2019-04-18:
Good morning !

I'm using the D435 mounted on a robot for 3D obstacle detection. But when the robot gets close to the edge of a new corridor, there is a veiling effect that creates obstacles points (see pictures below).

Is there an option or some post-processing filter that I missed and that can remove these points ? 

Thanks for your help

[image]
[image]

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

@Patrick-Lascombe Sorry for late response. Is this issue still existing for you? Could you please also provide one color image of your testing environment so that we can better understand this issue? Thanks!

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@Patrick-Lascombe  Didn't get response from your side for weeks. Will close the ticket at this point. Please feel free to create another one if you still have questions or issues. Thanks!

closed 2019-12-13 by RealSenseCustomerSupport