realsenseai/realsense-ros #738: Unable to set inter_came_sync_mode with default value

jasonbeach, 2019-04-18:
I'm trying to set `inter_cam_sync_mode = 1` when I start the ros node.  I saw issue https://github.com/intel-ros/realsense/issues/609
and the associated fix and have made the appropriate changes to the ros launch files.
I've verified that the parameter is set right and this line of code is executed with the `inter_cam_sync_mode` option:
https://github.com/intel-ros/realsense/blob/f548ca5e43229d22c75ac3f994f9e4d85debf4d8/realsense2_camera/src/base_realsense_node.cpp#L301

I query the value immediately before and after that statement and verified before setting it the value is 0 and after it is 1. However, I get no pulse coming out of pin 5.

If instead I add this line:
`sensor.as<rs2::depth_sensor>().set_option(RS2_OPTION_INTER_CAM_SYNC_MODE,1);`
immediately after this line
https://github.com/intel-ros/realsense/blob/f548ca5e43229d22c75ac3f994f9e4d85debf4d8/realsense2_camera/src/base_realsense_node.cpp#L1604

then it works as expected (obviously no changes are need to the launch files in this case).  What surprises me most is that the `setupStreams()` function where I make this change happens before the `registerDynamicOption` function does so it's not like the sensor isn't ready.  Is there any other reason the first method wouldn't work?

Garand0o0, 2019-06-20:
Hi jasonbeach,
I want to use 2 d435i.Master and slave.
camera1:inter_cam_sync_mode = 1
camera2:inter_cam_sync_mode = 2
How should I revise the launch file to set master and slave?
thanks
My launch:
```
<launch>

  <!-- Multi-cameras demo with 2 D435i -->

  <!-- Choose visualization -->
  <arg name="rviz"       default="false" />
  <arg name="rtabmapviz" default="true" /> 

  <!-- Cameras -->
  <include file="$(find realsense2_camera)/launch/rs_rgbd.launch">
    <arg name="camera" value="camera1" />
    <arg name="serial_no" value="843112071072" />
  </include>

  <include file="$(find realsense2_camera)/launch/rs_rgbd.launch">
    <arg name="camera" value="camera2" />
    <arg name="serial_no" value="838212071006" />
  </include>
</launch>
```

doronhi, 2019-07-03:
@jasonbeach , I looked into the case and it seems like this is a non-existent feature.
The inter_cam_sync_mode must be set before the sensor starts streaming, like fps, width, height etc.
The parameters given by rosparam are set after the sensor is already streaming and therefor they don't affect the device.
There should be added a new parameter to the launch files, for the initialization of the device, as other parameters do.

doronhi, 2019-07-11:
fixed in #843. Merged into version 2.2.7

closed 2019-09-13 by RealSenseCustomerSupport