realsenseai/realsense-ros #739: No IMU intrinsics published by T265 node

schmidtp1, 2019-04-19:
The node advertises `/camera/accel/imu_info` and `/camera/gyro/imu_info`
but no messages arrive,
for example:
```
$ rostopic hz /camera/accel/imu_info
subscribed to [/camera/accel/imu_info]
no new messages
no new messages
no new messages
```

doronhi, 2019-06-10:
Thanks. I add #809 but before merging, I have a doubt:
All the imu_info messages are the same so is it advisable to publish them as a topic or should we have a service for that? 
Any thoughts on the subject?

doronhi, 2019-06-11:
Add #810 - solution with services.

schmidtp1, 2019-06-11:
I would prefer them to be published to make sure to have them in the bag when recording.
Maybe they could be "latched"?

doronhi, 2019-06-12:
I like the idea. I fixed PR #809. What do you think?

closed 2019-10-04 by RealSenseCustomerSupport