realsenseai/realsense-ros #741: Is auto exposure region of interrest (AE ROI) exposed?

jdwolk, 2019-04-23:
Hi,

Is the auto exposure region of interest functionality exposed?

I'm having issues w/ direct sunlight like those described in https://github.com/IntelRealSense/librealsense/issues/2875. There they recommend setting the region of interest so that it won't include the sun. Is this functionality exposed in any way? I don't see it anywhere in the dynamic reconfig params

MartyG-RealSense, 2019-04-24:
The link below has a Python script for setting the auto-exposure ROI.

https://github.com/IntelRealSense/librealsense/issues/2681#issuecomment-436391721

RealSenseCustomerSupport, 2019-10-25:
----------------------------------------------

@MartyG-RealSense Thanks for the help!
@jdwolk Does the script work for you? Looking forward to your update!

doronhi, 2019-10-27:
What device are you using? When it is available, like in the D435, the option is exposed in the dynamic reconfigure params:
[image]

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@jdwolk Any other questions for this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@jdwolk  If no other questions about this issue, will close it accordingly. Thanks!

closed 2019-12-02 by RealSenseCustomerSupport

phufetch, 2020-07-17:
@doronhi does the settings (AE ROI) stay on camera after a power cycle ? Is there a way to query this value on the camera ?

Thanks,
-Phu

MartyG-RealSense, 2020-07-17:
Hi @phufetch  Does the C++ script in the link below help you to query ("get") the auto-exposure ROI, please?

https://github.com/IntelRealSense/librealsense/issues/4363#issuecomment-508405760

An AE ROI "get" script in Python would look like the script in this link:

https://github.com/IntelRealSense/librealsense/issues/2681#issuecomment-436391721

I would expect the ROI to return to default values after a power-cycling.  It cannot be set via a json configuration file.