realsenseai/realsense-ros #742: Any docs indicates available filters?

jinfagang, 2019-04-26:
I want set a detph filters in launch file:

```
  <arg name="filters"             default="hole_filling"/>

```

But this filter method actually does not exist.

How many filter methods exist now?

MartyG-RealSense, 2019-04-26:
Documentation for the post-processing filter types can be found here:

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md

This is the link on that page to the Holes Filling Filter section.

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#holes-filling-filter

jinfagang, 2019-04-26:
@MartyG-RealSense  Thanks Sir. Could u provide which string it allowed in realsense_ros that can set inside launch files?

From those docs I don't know which string mapping which method

MartyG-RealSense, 2019-04-26:
ROS launch parameters for the filters are listed in the section linked to below.

https://github.com/intel-ros/realsense#launch-parameters

**disparity** - convert depth to disparity before applying other filters and back.
**spatial** - filter the depth image spatially.
**temporal** - filter the depth image temporally.
**hole_filling** - apply hole-filling filter.
**decimation** - reduces depth scene complexity.

It seems that the launch format is to put the word **filters:=** and then the filter name that you want to launch.  If there is more than one filter name, separate them with commas.  For example:

roslaunch realsense2_camera rs_camera.launch **filters:=pointcloud**

zinan93, 2019-06-13:
Hi, how would I set the parameters in the launch file then? For example setting the decimation factor to 4? and also how to use disparity filter if I want to filter in the following sequence for the depth stream: decimation(4)->disparity transfrom->spatial->temporal->disparity->hole_filling ?

MartyG-RealSense, 2019-06-13:
This is not quite my area of expertise so I apologize if I am mistaken.  It looks from the link below though that you can set the general filter type in the roslaunch instruction, and then set the specifics of the filter in a 'rosrun' instruction after the roslaunch command, using dynamic reconfigure.

https://github.com/IntelRealSense/realsense-ros/issues/583

zinan93, 2019-06-26:
Okay I see, thanks. I ll try dynamic reconfigure.

adrianramospeon, 2019-07-22:
I've got the same question. Can I set the parameters for the filters in the launch file? If so, how?

doronhi, 2019-07-31:
@adrianramospeon ,
Check #817 ,
Does that works for you?

adrianramospeon, 2019-07-31:
@doronhi 
That works nicely for the emitter (and many other parameters). Unfortunately, I haven't seen the same way of setting the post-processing filtering parameters. Right now I'm using something like follows for each post-processing filter.
```
<node name="configure_temporal_filter" pkg="dynamic_reconfigure" type="dynparam"
	args="set_from_parameters camera/temporal">
    <param name="filter_smooth_delta" value="50" />
    <param name="filter_smooth_alpha" value="0.1"/>
    <param name="holes_fill"          value="3"  />
  </node>
```
It works, and I'm satisfied with it, but it's a different style and a bit messier than the rest of the parameters (like infra_width, depths_fps, etc...). I now understand that these post-processing parameters are meant to be fiddled with, while the of the others described in [](https://github.com/IntelRealSense/realsense-ros#launch-parameters) are rather static, and hence the difference. Thanks and thumbs up for your work on this project!

doronhi, 2019-08-06:
Thank you.
The last commit of #858 fixed that issue as well. You can change the initial values of the filters by adding the following code to your launch file:
```
It will be incorporated in the upcoming release.
  <rosparam>
      /camera/stereo_module/emitter_enabled: true
      /camera/stereo_module/enable_auto_exposure: true
      /camera/temporal/filter_smooth_delta: 50
  </rosparam>
```

RealSenseCustomerSupport, 2019-11-04:
----------------------------------------------

@adrianramospeon Any other questions about this? Looking forward to your update. Thanks!

adrianramospeon, 2019-11-04:
> @adrianramospeon Any other questions about this? Looking forward to your update. Thanks!

No further questions, previous reply by @doronhi provided exactly what I was looking for. Thanks for the great support. It might be useful for newcomers to add a couple of documentation lines about this in https://github.com/IntelRealSense/realsense-ros#launch-parameters ?

closed 2019-11-04 by jinfagang

eibalazs, 2020-10-26:
@doronhi 
Where can I find the list of parameter names? I tried to find out the filter parameter namings based on the realsense-viewer namings, but there are diffferences. Is there a page where parameters like "/camera/temporal/filter_smooth_delta" are listed?

doronhi, 2020-11-02:
The list of supported filters is available, as @MartyG-RealSense commented [above](https://github.com/IntelRealSense/realsense-ros/issues/742#issuecomment-487034341).
If you type: `rosrun dynamic_reconfigure dynparam list` you'll get a list of all the reconfigurable nodes, including the ones for the filters you have chosen.
For instance, when we add the "temporal" filter, the list we get is something like that:
```
/camera/rgb_camera
/camera/rgb_camera/auto_exposure_roi
/camera/stereo_module
/camera/stereo_module/auto_exposure_roi
/camera/temporal
```
To get a list of available parameters for the temporal filter run: `rosrun dynamic_reconfigure dynparam get /camera/temporal`
You then get the following list:
```
{'filter_smooth_alpha': 0.4000000059604645, 'stream_index_filter': -1, 'groups': {'filter_smooth_alpha': 0.4000000059604645, 'parent': 0, 'stream_index_filter': -1, 'groups': {}, 'id': 0, 'name': 'Default', 'parameters': {}, 'state': True, 'stream_format_filter': 1, 'stream_filter': 1, 'filter_smooth_delta': 20, 'type': '', 'holes_fill': 3, 'frames_queue_size': 16}, 'stream_format_filter': 1, 'stream_filter': 1, 'filter_smooth_delta': 20, 'holes_fill': 3, 'frames_queue_size': 16}
```
Where you can find the available parameters and their current value.

All those parameters are not mentioned in the realsense2_camera's documentation because they are inherited, in runtime, from the librealsense2's objects. The documentation regarding each filter is maintained in [librealsense2's documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md). Names were changed to follow ROS conventions.