realsenseai/realsense-ros #744: Ros wrapper for simultaneous cameras: D415 and T265

AdriannaGmz, 2019-04-27:
Hello!
I have a **D415** and a **T265** and need to access to their ros topics simultaneously. Currently working in Ubuntu 18, ROS melodic, although it seems to behave similarly in ubuntu 16 with ROS kinetic. So far I've compiled librealsense 2.21.0, 2.20 and 2.19 from source along with the ros wrapper 2.2.3 and here are the situations:

1. when entering `lsusb`, both cameras are sometimes recognized, sometimes just the D415 and sometimes just the T265. What I do, is unplugging them and connecting them back, trying my luck. this works say 80% of the times

2. when I execute `realsense-viewer`, both cameras are recognized  and display info, so the viewer seems to work fine

3. when listing devices:   ` rs-ennumerate-devices | grep Serial`       most of the times, only the D415 is recognized. Sometimes both cameras are listed, but that doesn't happen everytime, say 3/5 times

and finally, the **most important**:

4. when trying to run both cameras, they just don't work or one node dies. So far I've tried with the following launch files:

* roslaunch realsense2_camera rs_d400_and_t265.launch

* rs_multiple_devices.launch

And here are the outcomes from their executions

**roslaunch realsense2_camera rs_d400_and_t265_turtlebot.launch  (setting the serial number inside the launch file)**

```
... logging to /home/pidskull/.ros/log/04bca954-687f-11e9-88de-001fc69d0254/roslaunch-pidskull-desktop-20811.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pidskull-desktop:41295/

SUMMARY
========

PARAMETERS
 * /d415/realsense2_camera/accel_fps: 0
 * /d415/realsense2_camera/accel_frame_id: d415_accel_frame
 * /d415/realsense2_camera/accel_optical_frame_id: d415_accel_optica...
 * /d415/realsense2_camera/align_depth: False
 * /d415/realsense2_camera/aligned_depth_to_color_frame_id: d415_aligned_dept...
 * /d415/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d415_aligned_dept...
 * /d415/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d415_aligned_dept...
 * /d415/realsense2_camera/aligned_depth_to_fisheye_frame_id: d415_aligned_dept...
 * /d415/realsense2_camera/aligned_depth_to_infra1_frame_id: d415_aligned_dept...
 * /d415/realsense2_camera/aligned_depth_to_infra2_frame_id: d415_aligned_dept...
 * /d415/realsense2_camera/base_frame_id: d415_link
 * /d415/realsense2_camera/calib_odom_file: 
 * /d415/realsense2_camera/clip_distance: -1.0
 * /d415/realsense2_camera/color_fps: 30
 * /d415/realsense2_camera/color_frame_id: d415_color_frame
 * /d415/realsense2_camera/color_height: 480
 * /d415/realsense2_camera/color_optical_frame_id: d415_color_optica...
 * /d415/realsense2_camera/color_width: 640
 * /d415/realsense2_camera/depth_fps: 30
 * /d415/realsense2_camera/depth_frame_id: d415_depth_frame
 * /d415/realsense2_camera/depth_height: 480
 * /d415/realsense2_camera/depth_optical_frame_id: d415_depth_optica...
 * /d415/realsense2_camera/depth_width: 640
 * /d415/realsense2_camera/enable_accel: True
 * /d415/realsense2_camera/enable_color: True
 * /d415/realsense2_camera/enable_depth: True
 * /d415/realsense2_camera/enable_fisheye1: False
 * /d415/realsense2_camera/enable_fisheye2: False
 * /d415/realsense2_camera/enable_fisheye: True
 * /d415/realsense2_camera/enable_gyro: True
 * /d415/realsense2_camera/enable_infra1: True
 * /d415/realsense2_camera/enable_infra2: True
 * /d415/realsense2_camera/enable_pointcloud: False
 * /d415/realsense2_camera/enable_sync: False
 * /d415/realsense2_camera/filters: 
 * /d415/realsense2_camera/fisheye1_frame_id: d415_fisheye1_frame
 * /d415/realsense2_camera/fisheye1_optical_frame_id: d415_fisheye1_opt...
[... 431 lines cut ...]
 26/04 16:58:59,207 ERROR [140154682914560] (context.cpp:433) Exception thrown from user callback handler
 26/04 16:58:59,210 WARNING [140155632994176] (rs.cpp:279) null pointer passed for argument "device"
[ERROR] [1556323139.210296137]: An exception has been thrown: Could not open device. Last Error: Device or resource busy
[FATAL] [1556323139.210904694]: Failed to load nodelet '/camera1/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera1/realsense2_camera-3] process has died [pid 20973, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/pidskull/.ros/log/40d28026-687f-11e9-88de-001fc69d0254/camera1-realsense2_camera-3.log].
log file: /home/pidskull/.ros/log/40d28026-687f-11e9-88de-001fc69d0254/camera1-realsense2_camera-3*.log
^C[camera2/realsense2_camera-5] killing on exit
[camera1/realsense2_camera_manager-2] killing on exit
[camera2/realsense2_camera_manager-4] killing on exit
[ INFO] [1556323153.181146609]: Interrupt Signal is received! Terminating RealSense Node...
terminate called after throwing an instance of '[ INFO] [1556323153.181617945]: Interrupt Signal is received! Terminating RealSense Node...
rs2::error'
  what():  null pointer passed for argument "device"
 26/04 16:59:13,184 WARNING [139787587520256] (sensor.cpp:453) Frame received with streaming inactive,Depth0, Arrived,0.000000 1556323153184.201904
[camera2/realsense2_camera_manager-4] escalating to SIGTERM

```

for completion purposes, I've allowed myself to add the following rosbag when launching `rs_multiple_devices `

[2019-04-26-16-31-22.bag.zip](https://github.com/intel-ros/realsense/files/3123320/2019-04-26-16-31-22.bag.zip)

Could you **please** provide guidance on how to get both cameras working at the same time, publishing their respecting ROS topics?

Thanks in advanced

schmidtp1, 2019-04-29:
Hi @AdriannaGmz, thank you for reporting this issue!
I'm looking into it after I also observed at least one of the cameras crashing in ROS node frequently when using both cameras together.
But I don't see issues (1) lsusb and (3) rs-enumerate in my setup, librealsense 2.21.0 on Ubuntu 16.04 with T265 and D435i (FW versions below).
I tried un- and replugging multiple times and in different orders but can't reproduce. Didn't get to try on Ubuntu 18.04 (yet).
```
$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D435I         838212071360        05.11.00.09
255.255.255.255
Intel RealSense T265          845412110395        0.0.18.5502         
Device info: 
    Name                          : 	Intel RealSense D435I
    Serial Number                 : 	838212071360
    Firmware Version              : 	05.11.00.09
255.255.255.255
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B3A
    Usb Type Descriptor           : 	3.2

Device info: 
    Name                          : 	Intel RealSense T265
    Serial Number                 : 	845412110395
    Firmware Version              : 	0.0.18.5502
    Physical Port                 : 	vid_8087 pid_0B37 bus_2 port_2
    Product Id                    : 	0B37
```

AdriannaGmz, 2019-04-29:
hi @schmidtp1 ! thanks for commenting! yes, I'm testing it out in Ubuntu 18.04 precisely and yeah, here are the firmware versions as well

 ```
$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D415          806312060087        05.10.13.00         
Intel RealSense T265          845412111230        0.0.18.5502         
Device info: 
    Name                          : 	Intel RealSense D415
    Serial Number                 : 	806312060087
    Firmware Version              : 	05.10.13.00
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD3
    Usb Type Descriptor           : 	3.2

Device info: 
    Name                          : 	Intel RealSense T265
    Serial Number                 : 	845412111230
    Firmware Version              : 	0.0.18.5502
    Physical Port                 : 	vid_8087 pid_0B37 bus_2 port_2
    Product Id                    : 	0B37
```

schmidtp1, 2019-04-29:
About (2) realsense-viewer, when I start with both cameras connected, only one, D435i, shows up in the list on the left side. The second one, T265, has to be added as source manually (see screenshots below).
When un-/replugging the 2nd camera always has to be added as source manually.
[image]
[image]
[image]

AdriannaGmz, 2019-04-29:
True, it needs to be added manually, but when it's done, works ok for the gui. 

my real concern is about the ros wrapper and why both  devices are hard to reach by software. For instance, when , when executing the multiple_devices launch , get errors such as

```
[FATAL] [1556563879.319976002]: Failed to load nodelet '/camera2/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera2/realsense2_camera_manager-4] process has died [pid 19069, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/pidskull/.ros/log/c59159d6-6aaf-11e9-9793-001fc69d0254/camera2-realsense2_camera_manager-4.log].
log file: /home/pidskull/.ros/log/c59159d6-6aaf-11e9-9793-001fc69d0254/camera2-realsense2_camera_manager-4*.log
[camera2/realsense2_camera-5] process has died [pid 19070, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/pidskull/.ros/log/c59159d6-6aaf-11e9-9793-001fc69d0254/camera2-realsense2_camera-5.log].
log file: /home/pidskull/.ros/log/c59159d6-6aaf-11e9-9793-001fc69d0254/camera2-realsense2_camera-5*.log
 29/04 11:51:20,000 WARNING [139794910144256] (rs.cpp:279) null pointer passed for argument "device"
 29/04 11:51:20,001 ERROR [139794910144256] (context.cpp:433) Exception thrown from user callback handler
 29/04 11:51:20,015 WARNING [139795859957632] (rs.cpp:279) null pointer passed for argument "device"
[ERROR] [1556563880.015213324]: An exception has been thrown: Could not open device. Last Error: Device or resource busy
[FATAL] [1556563880.015811245]: Failed to load nodelet '/camera1/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera1/realsense2_camera-3] process has died [pid 19068, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/pidskull/.ros/log/c59159d6-6aaf-11e9-9793-001fc69d0254/camera1-realsense2_camera-3.log].
log file: /home/pidskull/.ros/log/c59159d6-6aaf-11e9-9793-001fc69d0254/camera1-realsense2_camera-3*.log

```

manomitbal, 2019-04-29:
Hi,

I am suffering from the same issue. I am running Ubuntu 16.04 with a combination of D435s and a T265. The T265 does not show a tag next to it in realsense-viewer (USB 3.2) like it does with the D435 and is never automatically recognized. Apart from the unreliable booting of the T265, there is an issue where a maximum of 3 cameras can be launched after which future launches give an error "Device or resources busy"  through the ros launch files.

AdriannaGmz, 2019-04-30:
In my case, D415 with T265 working simultaneously, the solution was to use the SDK version **2.19.1** 
although is not the latest version, this set up works in Ubuntu 16 and 18
previous to reverse the lrs version, make sure to uninstall all the other installations.

schmidtp1, 2019-05-01:
This PR solves the issue for me: https://github.com/intel-ros/realsense/pull/752
Please give it a try if you want and please confirm if it works for you. Thanks!

manomitbal, 2019-05-01:
#752 fixes this issue, I can confirm! Thanks @schmidtp1

schmidtp1, 2019-05-01:
Thanks, @manomitbal!

closed 2019-05-01 by schmidtp1