realsenseai/realsense-ros #746: T265 unreliable start

joekeo, 2019-04-29:
Hello,

I have been working with the t265 for the past couple of weeks. I have a problem in which every other time that I run my launch file, the realsense camera will not start properly.

I have added the hardware reset to my launch file
`<arg name="initial_reset"       default="true"/>` and `<arg name="initial_reset"            value="$(arg initial_reset)"/>`

But one out of two times I launch my node I have any of this errors:

'[ERROR] [1556529706.257730479]: An exception has been thrown: Character [`] is not valid as the first character in Graph Resource Name [`??
[FATAL] [1556529706.274446561]: Failed to load nodelet '/rs_t265/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
 
or

[ERROR] [1556530126.782646817]: An exception has been thrown: Character [?] is not valid as the first character in Graph Resource Name [??C
[FATAL] [1556530126.797875362]: Failed to load nodelet '/rs_t265/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'

Anyone has the same issue? Is there any way to fix this? I have updated my code to the latest release release at the moment '2.2.2'

Many thanks,
pepe

AdriannaGmz, 2019-04-29:
maybe unhelpful at all, but yeah, I reckon the same behaviour with me . Haven't been able to fix it or find a workaround...

schmidtp1, 2019-05-01:
I added a sync that seems to solve the issue for me: https://github.com/intel-ros/realsense/pull/752

schmidtp1, 2019-05-01:
Also see https://github.com/intel-ros/realsense/issues/733.

closed 2019-05-02 by schmidtp1

joekeo, 2019-05-03:
Many thanks for this update, it is working wonders

AndreV84, 2020-08-31:
@schmidtp1 
Hi, could you help with minor modification here https://github.com/IntelRealSense/librealsense/issues/7176#issuecomment-683603852 , please?