realsenseai/realsense-ros #748: Error using catkin_make on ubuntu 14 + ros indigo

Deastan, 2019-04-29:
Hello every one,

I worked on Ubunutu 14.04 with ROS indigo. The camera is the realsense d435.
Python version is 3.6.8, gcc 5.5. 
Following the answer of this discussion: https://github.com/IntelRealSense/librealsense/issues/1030, I supposed that everything should work on ubuntu 14 with ros indigo.

Do, I follow: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
which mean compiling from source for ubuntu 14. The output is:

`Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/librealsense2.so.2.21.0
-- Installing: /usr/local/lib/librealsense2.so.2.21
-- Installing: /usr/local/lib/librealsense2.so
-- Up-to-date: /usr/local/include/librealsense2
-- Installing: /usr/local/include/librealsense2/rs_advanced_mode.hpp
-- Installing: /usr/local/include/librealsense2/rsutil.h
-- Installing: /usr/local/include/librealsense2/rs.h
-- Installing: /usr/local/include/librealsense2/rs.hpp
-- Installing: /usr/local/include/librealsense2/rs_advanced_mode.h
-- Up-to-date: /usr/local/include/librealsense2/hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_processing.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_internal.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_sensor.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_device.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_context.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_options.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_pipeline.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_frame.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_record_playback.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_export.hpp
-- Installing: /usr/local/include/librealsense2/hpp/rs_types.hpp
-- Up-to-date: /usr/local/include/librealsense2/h
-- Installing: /usr/local/include/librealsense2/h/rs_advanced_mode_command.h
-- Installing: /usr/local/include/librealsense2/h/rs_types.h
-- Installing: /usr/local/include/librealsense2/h/rs_device.h
-- Installing: /usr/local/include/librealsense2/h/rs_option.h
-- Installing: /usr/local/include/librealsense2/h/rs_processing.h
-- Installing: /usr/local/include/librealsense2/h/rs_pipeline.h
-- Installing: /usr/local/include/librealsense2/h/rs_config.h
-- Installing: /usr/local/include/librealsense2/h/rs_record_playback.h
-- Installing: /usr/local/include/librealsense2/h/rs_sensor.h
-- Installing: /usr/local/include/librealsense2/h/rs_internal.h
-- Installing: /usr/local/include/librealsense2/h/rs_context.h
-- Installing: /usr/local/include/librealsense2/h/rs_frame.h
-- Installing: /usr/local/lib/cmake/realsense2/realsense2Targets.cmake
-- Installing: /usr/local/lib/cmake/realsense2/realsense2Targets-noconfig.cmake
-- Installing: /usr/local/lib/cmake/realsense2/realsense2Config.cmake
-- Installing: /usr/local/lib/cmake/realsense2/realsense2ConfigVersion.cmake
-- Installing: /usr/local/lib/pkgconfig/realsense2.pc
-- Installing: /usr/local/lib/librealsense-file.a
-- Installing: /usr/local/bin/rs-hello-realsense
-- Set runtime path of "/usr/local/bin/rs-hello-realsense" to ""
-- Installing: /usr/local/bin/rs-software-device
-- Set runtime path of "/usr/local/bin/rs-software-device" to ""
-- Installing: /usr/local/bin/rs-capture
-- Set runtime path of "/usr/local/bin/rs-capture" to ""
-- Installing: /usr/local/bin/rs-callback
-- Set runtime path of "/usr/local/bin/rs-callback" to ""
-- Installing: /usr/local/bin/rs-save-to-disk
-- Set runtime path of "/usr/local/bin/rs-save-to-disk" to ""
-- Installing: /usr/local/bin/rs-multicam
-- Set runtime path of "/usr/local/bin/rs-multicam" to ""
-- Installing: /usr/local/bin/rs-pointcloud
-- Set runtime path of "/usr/local/bin/rs-pointcloud" to ""
-- Installing: /usr/local/bin/rs-align
-- Set runtime path of "/usr/local/bin/rs-align" to ""
-- Installing: /usr/local/bin/rs-align-advanced
-- Set runtime path of "/usr/local/bin/rs-align-advanced" to ""
-- Installing: /usr/local/bin/rs-sensor-control
[... 262 lines cut ...]
/home/roboticlab14/catkin_ws_camera/src/realsense/realsense2_camera/src/base_realsense_node.cpp:331:26: error: could not convert ‘nh1.ros::NodeHandle::param<int>(option_name, option_value, option_value)’ from ‘void’ to ‘bool’
             if (nh1.param(option_name, option_value, option_value))
                          ^
make[2]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[1]: *** [realsense/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

I also tried to installed with pip using "pip3 install pyrealsense2 --user" but if I tried to import pyrealsense2 it crashed with the following error.. And also with the tutorial test as showing in the terminal output : 
`roboticlab14@roboticlab14:~/catkin_ws_camera$ python3 /home/roboticlab14/Documents/Git/librealsense/wrappers/python/examples/python-tutorial-1-depth.py
Traceback (most recent call last):
  File "/home/roboticlab14/Documents/Git/librealsense/wrappers/python/examples/python-tutorial-1-depth.py", line 9, in <module>
    import pyrealsense2 as rs
  File "/home/roboticlab14/.local/lib/python3.6/site-packages/pyrealsense2/__init__.py", line 2, in <module>
    from .pyrealsense2 import *
ImportError: /home/roboticlab14/.local/lib/python3.6/site-packages/pyrealsense2/pyrealsense2.cpython-36m-x86_64-linux-gnu.so: symbol _ZTVNSt7__cxx1115basic_stringbufIcSt11char_traitsIcESaIcEEE, version GLIBCXX_3.4.21 not defined in file libstdc++.so.6 with link time reference`

Do someone have some ideas to help me please ? I start to be mad with this issue...

Thank you very much for your future help.

Best,

Jonathan

doronhi, 2019-05-13:
Hi Jonathan,
I am sorry for the long delay and the frustration.
I didn't try this realsense2_camera version on indigo but based on the build log you published there really seems to be a problem: The param function returns bool on kinetic onward but void in Indigo. I don't know if there are any other changes to the interface.
This version does not officially supports Indigo...

Deastan, 2019-05-15:
No problem. Thank you for your feedback.

closed 2019-06-10 by doronhi

blocka23, 2019-09-04:
Were you able to create a workaround for this? I'm having a similar issue.

ravijo, 2019-09-25:
@blocka23: I am using ROS Indigo on Ubuntu 14.04 LTS. As expected, the `realsense_ros` throws the following error `error: could not convert ‘nh1.ros::NodeHandle::param(option_name, option_value, option_value)’ from ‘void’ to ‘bool’`. I checked out an older version and recompiled to make it work.

Please use version 2.2.1 as I did just now. More specifically, instead of executing the following command:
```
git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`
```

Please execute the following command:
```
git checkout 2.2.1
```

While compilation, it may show multiple `ddynamic_reconfigure` package. You need to delete `ddynamic_reconfigure`, which is inside `ros_ws/src`. Please remember not to delete `ddynamic_reconfigure` which is inside `ros_ws/src/realsense-ros`.

Cheers

blocka23, 2019-09-25:
Dear @ravijo 
Thank you very much for your response. I did what you suggested and used git checkout 2.2.1

Unfortunately, I'm still having errors when I try to catkin_make

```
/home/block/catkin_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:42:45: fatal error: manipulation_msgs/GraspPlanning.h: No such file or directory
compilation terminated.
make[2]: *** [moveit/moveit_ros/manipulation/move_group_pick_place_capability/CMakeFiles/moveit_move_group_pick_place_capability.dir/src/pick_place_action_capability.cpp.o] Error 1
make[1]: *** [moveit/moveit_ros/manipulation/move_group_pick_place_capability/CMakeFiles/moveit_move_group_pick_place_capability.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 87%] Linking CXX shared library /home/block/catkin_ws/devel/lib/python2.7/dist-packages/moveit_ros_planning_interface/_moveit_roscpp_initializer.so
[ 87%] Built target moveit_py_bindings_tools_python
[ 88%] Linking CXX shared library /home/block/catkin_ws/devel/lib/libmoveit_planning_scene_interface.so
[ 88%] Built target moveit_planning_scene_interface
[ 88%] Linking CXX shared library /home/block/catkin_ws/devel/lib/python2.7/dist-packages/moveit_ros_planning_interface/_moveit_robot_interface.so
[ 88%] Built target moveit_robot_interface_python
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

```
Do you have any suggestions on how to fix this?

ravijo, 2019-09-25:
Hi @blocka23

The error is generated from moveit. This is a kind of common issue. You should run catkin_make install and catkin_make repeatedly a few times. It should disappear with in 2-3 runs.

blocka23, 2019-09-26:
Hi @ravijo 
Thank you for your continued assistance.

I did as you suggested and ran `catkin_make install` and `catkin_make` several times. I was able to get to 100%, but it still says `Invoking "make -j8 -l8" failed` The error I'm getting is still this fatal error from moveit. Do you have any other suggestions?
`/home/block/catkin_ws/src/moveit/moveit_ros/manipulation/move_group_pick_place_capability/src/pick_place_action_capability.cpp:42:45: fatal error: manipulation_msgs/GraspPlanning.h: No such file or directory
compilation terminated.
make[2]: *** [moveit/moveit_ros/manipulation/move_group_pick_place_capability/CMakeFiles/moveit_move_group_pick_place_capability.dir/src/pick_place_action_capability.cpp.o] Error 1
make[1]: *** [moveit/moveit_ros/manipulation/move_group_pick_place_capability/CMakeFiles/moveit_move_group_pick_place_capability.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX executable /home/block/catkin_ws/devel/lib/moveit_setup_assistant/moveit_setup_assistant
/usr/bin/ld: warning: libboost_system.so.1.54.0, needed by /opt/ros/indigo/lib/libroscpp.so, may conflict with libboost_system.so.1.57.0
/usr/bin/ld: warning: libboost_program_options.so.1.54.0, needed by /opt/ros/indigo/lib/librviz.so, may conflict with libboost_program_options.so.1.57.0
[100%] Built target moveit_setup_assistant
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

ravijo, 2019-09-26:
Hi @blocka23

This is a bit surprising for me to see `manipulation_msgs`. Which version of ROS are you using? As per [official ROS page](http://wiki.ros.org/Distributions), ROS Indigo was released on July 22nd, 2014. The [CHANGELOG](https://github.com/ros-planning/moveit/blob/master/moveit_commander/CHANGELOG.rst) file of `moveit` says that it no longer depends on `manipulation_msgs` from Sep 23th, 2014.

Furthermore, the GitHub page of [manipulation_msgs](https://github.com/ros-interactive-manipulation/manipulation_msgs) where [GraspPlanning](https://github.com/ros-interactive-manipulation/manipulation_msgs/blob/hydro-devel/srv/GraspPlanning.srv) is defined says that the package in this repo is DEPRECATED.

I definitely suggest you update the `moveit`. On ROS Indigo `sudo apt-get install ros-indigo-moveit-full` should work, although some users have experienced difficulties while installing `ros-indigo-moveit-full` as mentioned [here](https://answers.ros.org/question/251168/unable-to-locate-package-ros-indigo-moveit-full/?answer=251173#post-id-251173). Anyway, you can always build `moveit` from the source. I believe that the proper installation of moveit should fix your error.

Lastly, if you still fail, you should build `manipulation_msgs` package by copying it in your ROS workspace. However, please note that this package is deprecated.

_PS: On a side note, you are having two versions of boost i.e, v1.54 and v1.57._