realsenseai/realsense-ros #753: Stream width / height not loaded from json preset

ftbmynameis, 2019-05-01:
Hi,
I have saved the Preset "Default" settings in the librealsense-viewers tool to a json file and now want to load these settings, when I start a D415 camera. Most importantly in these settings the resolution is set to 1280x720 for the stereo module. However, when I load this settings file the starting log tells me that the resolution for all cameras (depth, infra1, infra2, color) is 640x480 (the setting used as the default value in the roslaunch file). I would have expected that the parameters in the json file would overwrite the default ros launch arguments, but they don't. I can explicitly pass arguments for each parameter during launch, but that is not what I want and also not suitable for a whole list of settings..
The launch command I am using is: 
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud json_file_path:=/path/to/config/DefaultPreset_viewer.json
The output I get is:
`process[camera/realsense2_camera_manager-1]: started with pid [26589]
process[camera/realsense2_camera-2]: started with pid [26590]
[ INFO] [1556720305.554051893]: Initializing nodelet with 12 worker threads.
[ INFO] [1556720305.623042321]: RealSense ROS v2.2.3
[ INFO] [1556720305.623071489]: Running with LibRealSense v2.20.0
[ INFO] [1556720305.991802014]: getParameters...
[ INFO] [1556720306.125659194]: setupDevice...
[ INFO] [1556720306.535370487]: JSON file is loaded! (/fzi/ids/jost/no_backup/thesis/DefaultPreset_viewer.json)
[ INFO] [1556720306.535452778]: ROS Node Namespace: camera
[ INFO] [1556720306.535490342]: Device Name: Intel RealSense D415
[ INFO] [1556720306.535517186]: Device Serial No: 746112061047
[ INFO] [1556720306.535542302]: Device FW version: 05.10.03.00
[ INFO] [1556720306.535570147]: Device Product ID: 0x0AD3
[ INFO] [1556720306.535594647]: Enable PointCloud: On
[ INFO] [1556720306.535623056]: Align Depth: Off
[ INFO] [1556720306.535664383]: Sync Mode: On
[ INFO] [1556720306.536219916]: Device Sensors: 
[ INFO] [1556720306.536587274]: Stereo Module was found.
[ INFO] [1556720306.536662531]: RGB Camera was found.
[ INFO] [1556720306.536731526]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556720306.536765269]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556720306.536792278]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556720306.536821717]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556720306.536849785]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556720306.536877295]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1556720306.537157070]: setupPublishers...
[ INFO] [1556720306.542755365]: Expected frequency for depth = 30.00000
[ INFO] [1556720306.549968958]: Expected frequency for infra1 = 30.00000
[ INFO] [1556720306.553581051]: Expected frequency for infra2 = 30.00000
[ INFO] [1556720306.558365179]: Expected frequency for color = 30.00000
[ INFO] [1556720306.566254901]: setupStreams...
[ INFO] [1556720306.588899549]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1556720306.589756244]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1556720306.590315611]: infra2 stream is enabled - width: 640, height: 480, fps: 30
 01/05 16:18:26,592 WARNING [140537109464960] (backend-v4l2.cpp:1248) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 01/05 16:18:26,593 WARNING [140537109464960] (sensor.cpp:366) Unregistered Media formats : [ RW16 ]; Supported: [ ]
[ INFO] [1556720306.605619062]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1556720306.607017036]: insert Depth to Stereo Module
[ INFO] [1556720306.607201368]: insert Color to RGB Camera
[ INFO] [1556720306.607254510]: insert Infrared to Stereo Module
[ INFO] [1556720306.607290328]: insert Infrared to Stereo Module
[ INFO] [1556720306.625843751]: Add Filter: pointcloud
[ INFO] [1556720306.638707188]: num_filters: 1
[ INFO] [1556720306.638804510]: SELECTED BASE:Depth, 0
[ INFO] [1556720306.647870462]: RealSense Node Is Up!
`
The json file explicitly sets the desired stream resolution:
`    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"`

Now I am very confused if any parameters are actually taken into account or if maybe it is just an output issue?

Thank you.

MartyG-RealSense, 2019-05-01:
The most recent information I know about this is that in 2018, the JSON could not set FPS and resolution because the settings made by the file were overridden by the pipeline.  The rest of the JSON settings loaded fine.  Dorodnic the RealSense SDK Manager said they planned to adjust the pipeline but I do not know if it has been implemented.  

https://github.com/IntelRealSense/librealsense/issues/1229#issuecomment-368329980

ftbmynameis, 2019-05-02:
Thank you for the fast reply. 

What I struggle to understand is that no matter what I do the log of the ros node tells me on startup that the selected resolution for all 4 camera streams is 640x480 which by what the official tuning whitepaper says is not a good resolution. It would recommend a resolution of 1280x720.
 Is the node output information misleading or am I missing something?

MartyG-RealSense, 2019-05-02:
My apologies for the delay in a subsequent reply, as I wanted to research this subject carefully.  I believe that the set resolution may be related to the launch file settings, and that you can 'dynamically reconfigure' the settings directly after the launch file instruction using **rosrun** in combination with **rqt_reconfigure**

https://github.com/intel-ros/realsense#set-camera-controls-using-dynamic-reconfigure-params

ftbmynameis, 2019-05-05:
When I launch the camera with the same launch file / arguments (or any others for that matter) as in the link you have posted and then start rqt_reconfigure it firstly shows me much less options (I only get rgb_camera and stereo module, everything else like color, depth, infra1/2,.. is missing for me) and if I click on "rgb_camera" rqt does crash immediately. For the stereo module option it is fine.

doronhi, 2019-05-06:
Regarding the crashing rqt_reconfigure, it is still unresolved. I just re-opened issue #607.
Any additional information regarding this topic is more than welcome.
Regarding the JSON preset. realsense2_camera use it to initialize the device parameters, like stereo_module->Exposure and such. You can check that specific option with rqt_reconfigure, even without checkoing the rgb_camera option. 
The stream fps, width and height relate to which stream is opened. Although realsense-viewer stores the information regarding the stream it was using as well, that information is not used by realsense2_camera.

closed 2019-06-10 by doronhi