realsenseai/realsense-ros #754: How to loop file with rs_from_file.launch?

rje, 2019-05-02:
I recently recorded several rosbags using the intel realsense viewer and while they open and run as expected, it'd be very useful to be able to loop these files for continuous playback.

I tried updating the call to cfg.enable_device_from_file() in realsense_node_factory.cpp to pass true for the repeat_playback parameter, but playback continues to halt at the end of the file. Is there another change I need to make for this to work? (eventually I plan to clean it up with a parameter and submit a PR, but I'd like to see it working as a proof of concept first).

Thanks!

doronhi, 2019-06-10:
Sorry. I think it's an open issue in the code. If you ever figure it out, a PR will be very much appreciated.

k-maheshkumar, 2019-08-14:
Hey, I am also looking for such a feature, so I would like to know if there is any update on this issue?

RealSenseCustomerSupport, 2019-10-30:
----------------------------------------------

Hi @rje,

Would you like to try the latest librealsense and ROS wrapper? You can refer to below example to see how it can help you implement the PR in ROS.
https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

closed 2020-02-07 by RealSenseCustomerSupport