realsenseai/realsense-ros #756: Problem working with four d435

Hunk86, 2019-05-03:
Hello,

I've the issue that i didn't get the pointcloud of all four cameras after I started them. The problem is that all cameras try to start on the same time. I'm able to solve this Problem when i replug a camera after I didn't get the polygon. (Workaround)

My launch file is the same as 

https://github.com/intel-ros/realsense/blob/development/realsense2_camera/launch/rs_multiple_devices.launch

only with 4 cameras

I have two questions :
1. What was your line of thinking when someone want to use more then one camera ? Should I write an own ros wrapper for this ? 
2. Do you have an idea how i can solve this problem without replug the cameras ? ( please no script with sleep ;-) ) 

Now the output of the logs. Did you need more information inside the log fills ?
I'm using ubuntu 18.04 and my Kernel is : 4.15.0-47-lowlatency

SUMMARY
========

PARAMETERS
 * /camera_front_bottom/realsense2_camera/accel_fps: 0
 * /camera_front_bottom/realsense2_camera/accel_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/accel_optical_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/align_depth: False
 * /camera_front_bottom/realsense2_camera/aligned_depth_to_color_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/base_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/calib_odom_file: 
 * /camera_front_bottom/realsense2_camera/clip_distance: 6.0
 * /camera_front_bottom/realsense2_camera/color_fps: 15
 * /camera_front_bottom/realsense2_camera/color_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/color_height: 480
 * /camera_front_bottom/realsense2_camera/color_optical_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/color_width: 640
 * /camera_front_bottom/realsense2_camera/depth_fps: 15
 * /camera_front_bottom/realsense2_camera/depth_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/depth_height: 480
 * /camera_front_bottom/realsense2_camera/depth_optical_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/depth_width: 640
 * /camera_front_bottom/realsense2_camera/enable_accel: True
 * /camera_front_bottom/realsense2_camera/enable_color: True
 * /camera_front_bottom/realsense2_camera/enable_depth: True
 * /camera_front_bottom/realsense2_camera/enable_fisheye1: True
 * /camera_front_bottom/realsense2_camera/enable_fisheye2: True
 * /camera_front_bottom/realsense2_camera/enable_fisheye: True
 * /camera_front_bottom/realsense2_camera/enable_gyro: True
 * /camera_front_bottom/realsense2_camera/enable_infra1: True
 * /camera_front_bottom/realsense2_camera/enable_infra2: True
 * /camera_front_bottom/realsense2_camera/enable_pointcloud: False
 * /camera_front_bottom/realsense2_camera/enable_sync: True
 * /camera_front_bottom/realsense2_camera/filters: pointcloud
 * /camera_front_bottom/realsense2_camera/fisheye1_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/fisheye1_optical_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/fisheye2_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/fisheye2_optical_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/fisheye_fps: 30
 * /camera_front_bottom/realsense2_camera/fisheye_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/fisheye_height: 0
 * /camera_front_bottom/realsense2_camera/fisheye_optical_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/fisheye_width: 0
 * /camera_front_bottom/realsense2_camera/gyro_fps: 0
 * /camera_front_bottom/realsense2_camera/gyro_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/gyro_optical_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/infra1_frame_id: camera_front_bott...
 * /camera_front_bottom/realsense2_camera/infra1_optical_frame_id: camera_front_bott...
[... 494 lines cut ...]
[0m[ INFO] [1556788079.309636185]: Enable PointCloud: On[0m
[0m[ INFO] [1556788079.309643142]: Align Depth: Off[0m
[0m[ INFO] [1556788079.309650346]: Sync Mode: On[0m
[0m[ INFO] [1556788079.309979775]: Device Sensors: [0m
[0m[ INFO] [1556788079.310158794]: Stereo Module was found.[0m
[0m[ INFO] [1556788079.310199578]: RGB Camera was found.[0m
[0m[ INFO] [1556788079.310225173]: Motion Module was found.[0m
[0m[ INFO] [1556788079.310267543]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...[0m
[0m[ INFO] [1556788079.310284762]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...[0m
[0m[ INFO] [1556788079.310299161]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...[0m
[0m[ INFO] [1556788079.310312475]: (Pose, 0) sensor isn't supported by current device! -- Skipping...[0m
[0m[ INFO] [1556788079.310480180]: setupPublishers...[0m
[0m[ INFO] [1556788079.312661902]: Expected frequency for depth = 15.00000[0m
[0m[ INFO] [1556788079.335085552]: Expected frequency for infra1 = 30.00000[0m
[0m[ INFO] [1556788079.350111510]: Expected frequency for infra2 = 30.00000[0m
[0m[ INFO] [1556788079.363225456]: Expected frequency for color = 15.00000[0m
[0m[ INFO] [1556788079.379215934]: Start publisher IMU[0m
[0m[ INFO] [1556788079.381900206]: setupStreams...[0m
 02/05 11:07:59,383 WARNING [140523137537792] (sensor.cpp:366) Unregistered Media formats : [ UYVY ]; Supported: [ ]
[33m[ WARN] [1556787999.301279405]: If not further specified the default values for the dynamic reconfigurable parameters will be loaded.[0m
rocess[camera_side_left/realsense2_camera_manager-8]: started with pid [3917][0m
[1mprocess[camera_side_left/realsense2_camera-9]: started with pid [3919][0m
[1mprocess[camera_side_right/realsense2_camera_manager-10]: started with pid [3921][0m
[1mprocess[camera_side_right/realsense2_camera-11]: started with pid [3925][0m
[1mprocess[rviz-22]: started with pid [4066][0m

schmidtp1, 2019-05-10:
Hi @Hunk86,

Did you try this with the latest development? (https://github.com/IntelRealSense/realsense-ros/commit/b4b863d19b0a5c968fd882c81f03535259a7d5f2)

doronhi, 2019-05-20:
Is the problem solved by 

> Hi @Hunk86,
> 
> Did you try this with the latest development? ([b4b863d](https://github.com/IntelRealSense/realsense-ros/commit/b4b863d19b0a5c968fd882c81f03535259a7d5f2))

Is the problem solved?
If not, I added [PR 775](https://github.com/IntelRealSense/realsense-ros/pull/775) that is more resilient to conflicting devices. I tried it with 3 cameras and it worked for me.

Hunk86, 2019-05-23:
No the problem isn't solved. I've updated to :+1: 
 [ INFO] [1558624198.529168823]: Running with LibRealSense v2.22.0
[ INFO] [1558624198.540920248]: RealSense ROS v2.2.4

and get more errors:

` 23/05 17:10:12,529 ERROR [140455280654080] (types.h:243) xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
[ERROR] [1558624212.529618403]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
 23/05 17:10:12,529 ERROR [140455280654080] (context.cpp:433) Exception thrown from user callback handler
[ WARN] [1558624213.296061464]: Hardware Notification:Motion Module force pause,1.55862e+12,Error,Hardware Error
[ERROR] [1558624213.296204599]: Hardware Reset is needed. use option: initial_reset:=true
 23/05 17:10:14,380 ERROR [140454280554240] (types.h:243) xioctl(VIDIOC_DQBUF) failed for fd: 33 Last Error: No such device
 23/05 17:10:14,380 ERROR [140454280554240] (backend-v4l2.cpp:1230) xioctl(VIDIOC_DQBUF) failed for fd: 33 Last Error: No such device
[ WARN] [1558624214.380548636]: Hardware Notification:xioctl(VIDIOC_DQBUF) failed for fd: 33 Last Error: No such device,1.55862e+12,Error,Unknown Error
 23/05 17:10:14,401 ERROR [140455244650240] (types.h:243) Cannot open '/dev/video12 Last Error: No such file or directory
 23/05 17:10:14,401 ERROR [140455244650240] (global_timestamp_reader.cpp:152) Error during time_diff_keeper polling: Cannot open '/dev/video12 Last Error: No such file or directory
 23/05 17:10:14,410 ERROR [140454741923584] (types.h:243) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 23/05 17:10:14,410 ERROR [140454741923584] (error-handling.cpp:68) Error during polling error handler: get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 23/05 17:10:17,514 WARNING [140454704514816] (backend-hid.cpp:572) iio_hid_sensor: Frames didn't arrived within 5 seconds
 23/05 17:10:17,515 WARNING [140454712907520] (backend-hid.cpp:572) iio_hid_sensor: Frames didn't arrived within 5 seconds
[ERROR] [1558624217.539764693]: The device has been disconnected!
 23/05 17:10:17,574 ERROR [140455280654080] (types.h:243) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 23/05 17:10:17,574 ERROR [140455280654080] (sensor.cpp:260) An error has occurred while stop_streaming()!
 23/05 17:10:17,575 WARNING [140454976788224] (backend-hid.h:48) write_fs_attribute with 0 failed. The specified path /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0013/HID-SENSOR-200076.6.auto/iio:device5/buffer/enable is not valid
 23/05 17:10:17,575 WARNING [140454976788224] (backend-hid.cpp:666) HID set_power 0 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0013/HID-SENSOR-200076.6.auto/iio:device5/buffer/enable
 23/05 17:10:17,576 WARNING [140454985180928] (backend-hid.h:48) write_fs_attribute with 0 failed. The specified path /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0013/HID-SENSOR-200073.5.auto/iio:device1/buffer/enable is not valid
 23/05 17:10:17,576 ERROR [140455280654080] (types.h:243) Failed to enable_sensor /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0013/HID-SENSOR-2000e1.7.auto/enable_sensor Last Error: No such file or directory
 23/05 17:10:17,576 ERROR [140455280654080] (sensor.cpp:805) An error has occurred while stop_streaming()!
 23/05 17:10:17,576 WARNING [140454985180928] (backend-hid.cpp:666) HID set_power 0 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0013/HID-SENSOR-200073.5.auto/iio:device1/buffer/enable
 23/05 17:10:17,576 WARNING [140455280654080] (backend-hid.cpp:614) open() failed!
 23/05 17:10:17,581 WARNING [140455280654080] (backend-hid.cpp:614) open() failed!
 23/05 17:10:17,586 WARNING [140455280654080] (backend-hid.cpp:614) open() failed!
 23/05 17:10:17,591 WARNING [140455280654080] (backend-hid.cpp:614) open() failed!
 23/05 17:10:17,596 WARNING [140455280654080] (backend-hid.cpp:614) open() failed!
 23/05 17:10:17,601 WARNING [140455280654080] (backend-hid.cpp:614) open() failed!
 23/05 17:10:17,607 WARNING [140455280654080] (backend-hid.cpp:614) open() failed!
 23/05 17:10:17,612 WARNING [140455280654080] (backend-hid.cpp:614) open() failed!
 23/05 17:10:17,617 WARNING [140455280654080] (backend-hid.cpp:614) open() failed!
 23/05 17:10:17,622 ERROR [140455280654080] (types.h:243) open() failed with all retries! Last Error: No such file or directory
`
your branch is already merged or ? Then i have to use the dev branch ?

Hunk86, 2019-05-24:
And which Backend are you using ?

I found some hacks here :

doronhi, 2019-06-10:
My kernel version is:
`
4.15.0-47-generic
`
I don't understand the question about the backends. Could you clarify?
Rereading your original post I see now that I didn't check it with Ubuntu18.04 and also, not with 3 D435i but with 2 and a T265. Maybe that's a significant difference.
I'll let you know when I found something useful.

Hunk86, 2019-06-12:
If you are using the flag -DFORCE_LIBUVC=true ?

I've updated the system to 
ros-realsense: 2.2.6
librealsense: v2.23.0

I've tested the setup now with different number of cameras. With 2 it is working, but I receive the limit on 4 cameras.

At the moment I get an exception rs2::invalid_value_error

[image]

When we start the 4 cameras with sleep between the launch file it is working

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

Hi @Hunk86,

Please provide us with the following details about your system configuration :
FW version on each camera
ROS distribution version
HW configuration details (platforms names, how the camera(s) is/are connected, etc)

Thank you!

RealSenseCustomerSupport, 2019-11-08:
----------------------------------------------

@Hunk86,

Do you still need help with this question?

closed 2019-11-14 by RealSenseCustomerSupport

Vivoslespagnos, 2019-11-29:
Hi, I'm on Ubuntu 18.04(LTS), Kernel 5.0.0-36-generic x86_64, ROS Melodic and i have the same error message about UVC data from "ds5-timestamps.cpp:64". 
I dowloaded Realsense2.0 SDK last upgrade : https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md &&  https://github.com/intel-ros/realsense#step-3-install-intel-realsense-ros-from-sources 

But it's steel doesn't work ! I don't know why ! 

thx u