realsenseai/realsense-ros #757: Occur an error when using multiple realsense

KentaKawamata, 2019-05-04:
Hi,
Launch two or more D series, or D series and T265 simultaneously causes an error.

## System setup

|||
|:--:|:--:|
|Ubuntu|16.04.5|
|kernel|4.15.0-47-generic|
|ROS| ros-kinetic|
|T265 firmware|0.0.18.5448|
|librealsense|2.21.0|
|ros wrapper|2.2.3|

## Execution and result

The following error is when T265 is launched after launching D435 with `rs_camera.launch`.
```
$ roslaunch realsense2_camera rs_t265.launch serial_no:=<serial number>  name:=T265
```

```
... logging to /root/.ros/log/e7c47f7c-6e71-11e9-a996-302432bdadda/roslaunch-kawa-17057.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kawa:41525/

SUMMARY
========

PARAMETERS
 * /T265/realsense2_camera/accel_fps: 62
 * /T265/realsense2_camera/accel_frame_id: T265_accel_frame
 * /T265/realsense2_camera/accel_optical_frame_id: T265_accel_optica...
 * /T265/realsense2_camera/align_depth: False
 * /T265/realsense2_camera/aligned_depth_to_color_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_fisheye1_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_fisheye2_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_fisheye_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_infra1_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/aligned_depth_to_infra2_frame_id: T265_aligned_dept...
 * /T265/realsense2_camera/base_frame_id: T265_link
 * /T265/realsense2_camera/calib_odom_file: 
 * /T265/realsense2_camera/clip_distance: -2.0
 * /T265/realsense2_camera/color_fps: 30
 * /T265/realsense2_camera/color_frame_id: T265_color_frame
 * /T265/realsense2_camera/color_height: 480
 * /T265/realsense2_camera/color_optical_frame_id: T265_color_optica...
 * /T265/realsense2_camera/color_width: 640
 * /T265/realsense2_camera/depth_fps: 30
 * /T265/realsense2_camera/depth_frame_id: T265_depth_frame
 * /T265/realsense2_camera/depth_height: 480
 * /T265/realsense2_camera/depth_optical_frame_id: T265_depth_optica...
 * /T265/realsense2_camera/depth_width: 640
 * /T265/realsense2_camera/enable_accel: True
 * /T265/realsense2_camera/enable_color: True
 * /T265/realsense2_camera/enable_depth: True
 * /T265/realsense2_camera/enable_fisheye1: True
 * /T265/realsense2_camera/enable_fisheye2: True
 * /T265/realsense2_camera/enable_fisheye: True
 * /T265/realsense2_camera/enable_gyro: True
 * /T265/realsense2_camera/enable_infra1: True
 * /T265/realsense2_camera/enable_infra2: True
 * /T265/realsense2_camera/enable_pointcloud: False
 * /T265/realsense2_camera/enable_sync: False
 * /T265/realsense2_camera/filters: 
 * /T265/realsense2_camera/fisheye1_frame_id: T265_fisheye1_frame
 * /T265/realsense2_camera/fisheye1_optical_frame_id: T265_fisheye1_opt...
[... 61 lines cut ...]
13:39:12.888 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.898 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.908 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.918 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.929 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.939 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.949 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.960 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.970 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.980 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:12.991 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:13.001 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:13.011 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:13.022 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:13.032 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
[ERROR] [1556977153.057096920]: The requested device with serial number 845412110931 is NOT found!
13:39:13.533 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:14.033 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:14.534 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:15.035 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:15.536 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:16.037 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:16.537 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:39:17.038 [17121] [E] Device-7FE0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
```

KentaKawamata, 2019-05-06:
Sorry, the cause of this problem was in my environment.

I solved this problem.

closed 2019-05-06 by KentaKawamata

felix-v, 2019-05-09:
May I ask how you solved this problem?

HeinzBenjamin, 2019-11-21:
@KentaKawamata It would be really useful to know HOW you resolved the error.