realsenseai/realsense-ros #761: building ROS wrapper: internal compiler error: segfault

pringithub, 2019-05-08:
Hello, 

I am trying to get my RPi3B+ to run a ROS node and stream data over the network. I have an image with Ubuntu 16.04 + ROS Kinetic preinstalled. I have downloaded and installed the SDK (possibly related problem: upon launching the realsense-viewer, I was met with a segfault - attributing this to OpenGL madness, I dismissed it as a GUI issue (i don't need a GUI) ). Upon setting up my catkin_ws and running the  catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release command, I am greeted with the following 

/usr/include/boost/type_traits/is_function.hpp:50:35: internal compiler error: Segmentation fault
     template< typename T > struct result_
                                   ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
realsense-ros-2.2.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense-ros-2.2.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense-ros-2.2.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1451: recipe for target 'realsense-ros-2.2.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense-ros-2.2.3/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2" failed

doronhi, 2019-05-12:
How did you get the SDK? Did you build it yourself? If not you should try it, in case there is a mismatch between the installation and the system.
Can you run `rs-enumerate-devices`? What is the output?

pringithub, 2019-05-14:
I believe one of the problems was not having the correct time on my rpi - I must have missed the warnings while building in catkin_make. (perhaps escalate it to an error? better an error than a segfault with no comment or backtrace ability)

closed 2019-05-14 by pringithub