realsenseai/realsense-ros #762: Sampling rate with copy vs linear_interpolation

IoannisKaragiannis, 2019-05-10:
Hi,

I'm quite new with realsense-ros. For now I'm just trying to verify that the sampling frequencies are as they are supposed to be. By using the default accel_fps(=250Hz) and gyro_fps(=400Hz), and while recording some data in a rosbag I estimated the sampling rate to be 253.82Hz and 400.227Hz for the accelerometer and gyroscope samples respectively. So far so good.

I then tried to launch the realsense camera by using the unite_imu_method. My estimated sampling frequency from the recorded data while listening to the camera/imu topic and storing them in a rosbag, was 507.57Hz and 654.21Hz for the linear_interpolation and the copy method respectively.

I saw in some older issues how the copy methods works and I'm not that interested. However, I want to understand why you picked linear_interpolation? For me there are two resampling methods to sync the data from two sensors:
a) Upsample (linear_interpolation) both to a higher common rate
b) Downsample the faster one to fit the rate of the slower one

With the first one we add extra information to both signals. With the second one you only touch one of the two signals, you loose information w.r.t. the faster one, and I think it's more efficient compared to the upsamling method.

Could you please enlighten me why you decided that (a) is better than (b)? Or direct me to some relative literature.

Thanks in advance for your support.

doronhi, 2020-05-24:
Hi,
I think you might be using an old version. Currently the published rate for both "copy" and "linear_interpolation" is the gyro's rate (around 400 Hz by default in rs_camera.launch file).
The information in the README.md file is old, misleading, and I'll fix that with the following explanation:
"copy" - every gyro message is attached by the last accel message.
"linear_interpolation" - every gyro message is attached by the an accel message interpolated to the gyro's timestamp.

The true information is the original information, coming from the device on 2 different topics and in the original frequencies. Combining them in this wrapper is meant to ease the use with popular navigation nodes. A better way to combine them could involve some kalman filtering or other methods. That could be done using other ROS nodes or by the specific user's application, but is considered to be beyond the scope of this wrapper.

RealSenseCustomerSupport, 2020-05-29:
----------------------------------------------

@IoannisKaragiannis Any other questions with doronhi's reply? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-06-22:
----------------------------------------------

@IoannisKaragiannis  Sorry that we didn't get response from you for weeks. Will close this accordingly.

closed 2020-06-22 by RealSenseCustomerSupport