realsenseai/realsense-ros #763: Intel t265: where is IMU frame?

PerrineAguiar, 2019-05-10:
Hi,

In which frame is supposed to be expressed the raw IMU data?

Regarding the orientation, none of the frames published by the driver seems to correspond to the actual IMU frame. I would expect that `camera_accel_frame` or `camera_accel_optical_frame` should be the one, but looking at the raw IMU data, the actual IMU frame should have: x pointing forward, y pointing right and z pointing down.

Regarding the position, is the IMU located at the origin of the `camera_accel_frame` or at the origin of the `camera_pose_frame`?

Thanks in advance.

schmidtp1, 2019-05-10:
Hi @PerrineAguiar,

The raw sensor readings are expressed in the `camera_accel_optical_frame` but then transformed: https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L1245
In case the target frame should be `camera_accel_frame`, the rotation matrix (to transform from optical frame to frame) should be `[0 0 1; -1 0 0; 0 -1 0]`.
Personally, I think the data should just be expressed in the original frame.

The origin of the IMU is located at `camera_accel_frame`. (The `camera_pose_frame` is centered between the two fisheye.)

PerrineAguiar, 2019-05-14:
@schmidtp1 thanks for the quick answer.
Regarding the orientation, there might be a bug. If I look at the the `camera_accel_frame` and the `camera_accel_optical_frame` in rviz, it looks like the conversion function should be something like:
```
void BaseRealSenseNode::ConvertFromOpticalFrameToFrame(float3& data)
{
    float3 temp;
    temp.x = -data.z;
    temp.y = -data.x;
    temp.z = data.y;

    data.x = temp.x;
    data.y = temp.y;
    data.z = temp.z;
}
```

doronhi, 2019-05-16:
@PerrineAguiar 
I think you may be right and there is a problem with that code. The solution, however, is not just fixing the function as you say, since it is also used by D435i.
I am working on it these days.

doronhi, 2019-05-21:
Fixed in #778 .
I agree with @schmidtp1 , removed the function ConvertFromOpticalFrameToFrame and changed the frame_id to be in the original coordinate system, as described in the _optical_frame.

closed 2019-06-10 by doronhi