realsenseai/realsense-ros #765: Partial point cloud data separation target object

tianjincb, 2019-05-11:
Observing the data in rviz, part of the data is separated from the target object itself and cannot be removed. What is the reason? How should I do?
[image]
[image]

tianjincb, 2019-05-11:
Use D435

doronhi, 2019-06-10:
Sorry for the long long delay..
Is this still relevant?
If so, I'll need more clarifications. I don't understand what is the problem I should see in the images.

RealSenseCustomerSupport, 2019-11-04:
----------------------------------------------

@tianjincb Is the issue still existing? If so, could you please clarify the issue details as doronhi requested? Thanks!

RealSenseCustomerSupport, 2019-12-02:
----------------------------------------------

@tianjincb Sorry that we didn't get feedback from you for weeks. Will close it at this point. Please feel free to create another one if you still have questions/issues. Thanks!

closed 2019-12-02 by RealSenseCustomerSupport