realsenseai/realsense-ros #769: Unable to play bag files with rs_from_file.launch

MCSuperior, 2019-05-14:
Hi there!

Maybe I do something wrong (I am not that experienced with RealSense), but it seems that .bag files which were created using the realsense-viewer cannot be played using `rs_from_file.launch`.

I am using
RealSense ROS v2.2.3
LibRealSense v2.21.0
Intel RealSense D435
Ubuntu 16.04

If I create a file `20190514_210535.bag` with the realsense-viewer and try to run it using `roslaunch realsense2_camera rs_from_file.launch rosbag_filename:="20190514_210535.bag"` (within the folder where the file is contained), I get the following error messages:

```
ERROR [139753262786432] (types.h:243) Failed to create ros reader: Error opening file: 20190514_210535.bag
ERROR [139753262786432] (types.h:243) Failed to create ros reader: Error opening file: 20190514_210535.bag
ERROR [139753262786432] (types.h:243) Failed to create ros reader: Error opening file: 20190514_210535.bag
WARNING [139753262786432] (rs.cpp:1557) null pointer passed for argument "profile"
[ERROR] [1557835631.230786127]: An exception has been thrown: Failed to resolve request. Request to enable_device_from_file("20190514_210535.bag") was invalid, Reason: Failed to create ros reader: Error opening file: 20190514_210535.bag
[FATAL] [1557835631.231047571]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
```

It seems as if the file is somehow wrong, but it came directly out of the realsense-viewer. Also, as far as I understood, stating the .bag file name in the command should be sufficient, right?

MCSuperior, 2019-05-15:
Turns out that the entire path is required for the filename. My mistake, sorry for bothering.
For simplifying the debug, however, I suggest to add a ROS_ERROR or something into the `void RealSenseNodeFactory::onInit()` which states if a file name is invalid. Something like this:

```
std::ifstream infile(rosbag_filename);
if (!infile.good())
{
    ROS_ERROR_STREAM("ERRONEOUS FILE NAME");
}
```

closed 2019-05-15 by MCSuperior

doronhi, 2019-05-16:
Note taken. Sorry for your inconvenience.

madhup620, 2020-04-11:
I faced the same issue in the following setting:
Windows 10 + Spyder 3 IDE + Python 3.7.4 + OpenCV

First I create a .bag file from a python file which recorded everything the camera captured. The file is placed in the same (expected) path as that of the script.

When I validate the existence of the file using os.path.exists, I get the correct result about whether the given file exists or not. Even then I get the same "ros' error.

Ofcourse, when I gave the FULL absolute path, os.path.exists tells the corrrect result and the 'ros' error does not occur.

Why is this specific limitation to use absolute path ? Is there any way to avoid it ?

thanks
Madhu