realsenseai/realsense-ros #772: T265 frame questions

Martin-Oehler, 2019-05-16:
Hello everyone,

I am currently trying to integrate the T265 into our mobile robot. However, I do have some problems understanding the meaning of the different TF frames published by the driver.

1. Position of the tracking frame
As I understand it, the frame [camera_name]_pose_frame refers to the tracking frame of the camera which is located in the center between left and right imager with X pointing forward, Y to the left and Z upwards. But I can't find information on the distance to the front plate. Is it located in the center of the housing (i.e. 6.25mm from the front plate)?

2. Position of the imagers:
When looking at the image streams we can see that fisheye1 is the left imager and fisheye1 the right. When I now display the fisheye1_optical_frame relative to the camera_pose_frame I would expect it to be located in positive Y direction. However, it is located in negative Y direction. Are the cameras switched in TF?

[image]

3. Fisheye camera frames
In ROS standard, the X axis of the camera frame should point forward and in the same direction as the Z-axis of the optical frame (see http://www.ros.org/reps/rep-0103.html#coordinate-frame-conventions). In the current TF tree, the X-axis of the fisheye camera frames point backwards.

4. Camera_link
A static camera link is published relative to the camera pose frame. What is the meaning of this frame? Where is it located and why is it rotated by 180° around Z? I expected this to be the center of the housing, but the transformation does not match the values given in the datasheet.

5. Odom calibration file
I want to integrate wheel odometry. As I understand it, the relative transformation to the sensor frame has to be given w.r.t the body frame. What exactly is the body frame? Is it the tracking frame, the center of the housing or something else?

I hope, you can clarify these points for me. Thank you for your help.

schmidtp1, 2019-05-16:
Hi @Martin-Oehler,

Thanks for your questions!

1. Please refer to the data sheet. The distance to the front plate should be 6.55mm.
[image]

2. That was a bug in librealsense that had the effect that the cameras appeared to be swapped. It was fixed in this PR: https://github.com/IntelRealSense/librealsense/pull/3900 that was just merged into the development branch. You can install from source and try it again.

3. I'll have to check. I was paying more attention to the optical frames which comply with the convention:

Martin-Oehler, 2019-05-16:
Thank you for your answers!

1. I see, I did not make the connection between IMU and tracking reference frame.

2. I will build from source, test again and report back.

3. Yes, the optical frames are indeed correct, so this is probably a minor issue.

4. No, there is a translation between camera_pose_frame and camera_link

```
rosrun tf tf_echo camera_pose_frame camera_link
At time 0.000
- Translation: [-0.000, 0.021, -0.000]
- Rotation: in Quaternion [0.000, 0.000, 1.000, -0.000]
            in RPY (radian) [-0.000, -0.000, -3.142]
            in RPY (degree) [-0.000, -0.000, -180.000]
```
5. Thanks, it's working now.

schmidtp1, 2019-05-16:
1. Please note that the frame is still centered between the 2 fisheye but all the sensors are in one plane (/ on one line).

4. I'll have to check that. Looks like the difference between center of fisheyes and IMU.

5. Great to hear!

doronhi, 2019-05-19:
Hi,
4) The camera_link frame was supposed to be some center point for all the other frames to relate to. In the D400 series it is defined as the depth frame. Here it is set as the Gyro frame, as this is a known physical position.

Martin-Oehler, 2019-05-20:
> That was a bug in librealsense that had the effect that the cameras appeared to be swapped. It was fixed in this PR: IntelRealSense/librealsense#3900 that was just merged into the development branch. You can install from source and try it again.

I confirm that this fixed the problem.

> Here it is set as the Gyro frame, as this is a known physical position.

I see. However, tf gives me the following transformation
```
$ rosrun tf tf_echo camera_fisheye1_frame camera_gyro_frame
At time 0.000
- Translation: [0.000, -0.011, 0.000]
- Rotation: in Quaternion [1.000, 0.004, 0.000, 0.001]
            in RPY (radian) [3.139, -0.001, 0.009]
            in RPY (degree) [179.841, -0.056, 0.503]
```

while the data sheet specifies a distance of 10 mm.

RealSenseCustomerSupport, 2019-08-26:
----------------------------------------------

Hi @Martin-Oehler

Quite a bit of updates to our T265 product has happened since this issue was posted.  If there is anything further necessary please outline from what you see in the latest librealsense release.  We should close this issue out if nothing further is necessary.

Thanks

Martin-Oehler, 2019-08-26:
I think all points have been addressed, thanks again.

closed 2019-08-26 by Martin-Oehler

dag08, 2022-11-22:
Hi, I have a problem with the measure of the tracking center. I measured it out like shown in the figure 4-2 and set it as nodal point deviation in unreal engine but it seams not to be the right spot. My question is from what a point should I measure it? I hope you understand what I mean.