realsenseai/realsense-ros #773: ROS openni2

BYEONGGILYOO, 2019-05-17:
Where do I add the librs2driver.so library to run realsense using ros openni2?
I have confirmed that realsense works in the openni2 sample(SimpleViewer).

Ubuntu 16.04
ROS Kinetic

MartyG-RealSense, 2019-05-17:
The link below has a user-contributed guide to setting up OpenNI2 and what to do with the librs2driver.so file once you have it (in step 12 of the guide).

https://github.com/IntelRealSense/librealsense/issues/2697#issuecomment-475042084

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

Hi @bgyoo91,

Has the link provided by @MartyG-RealSense helped to find an answer to your question? Please let us know if you need any further help with this matter.

Thank you!

closed 2019-11-06 by RealSenseCustomerSupport