realsenseai/realsense-ros #774: T265 and D435 launch issue. Failed to claim USB interface

urock, 2019-05-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 + D435  | 
| D435 Firmware Version                   | 05.11.04.00  | 
| LibRealSense Version                   |  2.21.0   |
| RealSense ROS Version                   |  2.2.3   |
| Operating System & Version |  Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC  |

### Issue Description

Hi everybody!
When launching with correct T265 and D425 serial numbers 
`roslaunch realsense2_camera rs_d400_and_t265.launch`
on every launch I get error
`Device-5880: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_E
RROR_BUSY)`
further however sometimes errors are gone, launch is ok for both cameras and I get pose from T265 and pointcloud from D435 topic streaming. 

But there is some percerntage of launches that would fail with additional error
`The requested device with serial number 845412110728 is NOT found!` 
845412110728 is my T265 module. 
In these cases launch keeps outputing error msgs and I get no topics from either camera. 

I am aware of [pull_request_752](https://github.com/IntelRealSense/realsense-ros/pull/752) (using latest development branch)

and I have followed setup instructions 
https://github.com/IntelRealSense/librealsense/blob/development/doc/installation.md#prerequisites
especialy I have made
`./scripts/setup_udev_rules.sh`
`./scripts/patch-realsense-ubuntu-lts.sh`

doronhi, 2019-05-19:
There is a problem where the node holds on to the context of the T265 device, even if not using it. That the reason why if the first node to start is T265 everything is ok but if the D400 is the first one up, the next node can't get a hold of the T265 context.
I am working on this issue right now.

akirayou, 2019-05-20:
I found a workaround.
Try to launch with nodelet "standalone" not "load".
It runs.

May be thread problem is there.

Here is my experimental patch for this.But still work in progress.
This patch  works, but some error message is still exists
(You can just ignore , error state will recover automatically with initial_reset)

https://github.com/akirayou/realsense/commit/6f5d8976b26c6428689a2f0cda90708f1cc63a1f
(tested on Raspi2/Ubuntu mate18 , librealsense with DFORCE_LIBUVC=false)

doronhi, 2019-05-20:
I submitted [PR 775](https://github.com/IntelRealSense/realsense-ros/pull/775)

urock, 2019-05-20:
Hi, guys 
Good to know you are aware of this issue. So this is realsense-ros wrapper issue and not librealsense?
@doronhi until your PR is merged I can suggest my own workaround. 

The trick is to delay launching D435 camera. Suppose you have t265.launch and d435.launch, you can use 

[timed_roslaunch](http://wiki.ros.org/timed_roslaunch) node to delay launching d435.launch like this

```
 <include file="$(find fastsense_px4)/launch/t265.launch">
    <arg name="camera1" value="$(arg camera1)" />
    <arg name="serial_no_camera1" value="$(arg serial_no_camera1)" />
  </include>

  <include file="$(find timed_roslaunch)/launch/timed_roslaunch.launch">
    <arg name="time" value="2" />
    <arg name="pkg" value="fastsense_px4" />
    <arg name="file" value="d435.launch" />
    <arg name="value" value="camera2:=$(arg camera2) serial_no_camera2:=$(arg serial_no_camera2)" />
  </include> 
```

akirayou, 2019-05-21:
> I submitted [PR 775](https://github.com/IntelRealSense/realsense-ros/pull/775)

It's good for recovering from the problem.Thank you.

But ,I want to point out another problem.
In the default launch file , T265 and D400 will launch on same nodelet.( yes, It's natural way)

But there is global variables such as "_device".
https://github.com/IntelRealSense/realsense-ros/blob/b286e4c3c4b55f4f1b8bb5aeaf1e65636677d4a0/realsense2_camera/src/realsense_node_factory.cpp#L22

So "_device" will be shared  by T265's RealSenseNodeFactory and D400's RealSenseNodeFactory.
I think It's not good. Static member is also danger, by same reason. 
(Except mutex and so on,which is need to share T265 and D400 device)

And additional small topic.
You don't have to handle signal by your self (i.e. C-c).
Nodelet will handle it , Only you have to do is close device on your destructor.
It's easy way to keep _device variable  in class member.you do not have to put _device in global.

doronhi, 2019-05-22:
@akirayou 
Thanks for the insight. It certainly makes things simpler!
I implemented and added into #778 as it includes #775.

akirayou, 2019-05-22:
@doronhi  Than you for reimplementing.
And I think you also should consider about static member in base_realsense_node.cpp

For example "seq"  in L1126  dose not have to be shared with other device.
And I'm not sure but, imu_history,init_gyro,init_accel, accel_factor,warn_count are neither.
Please inspect it.

These variables dose not make crash,just change the out put value.
So I can not test them well yet.

Ref:
https://github.com/akirayou/realsense/commit/6f5d8976b26c6428689a2f0cda90708f1cc63a1f#diff-15761e8f7813869ae417a96398deda78

doronhi, 2019-05-22:
I added `unite_imu_method:=linear_interpolation` to rs_multiple_devices.launch and ran 2 d435i cameras.
I compared the topics:
```
rostopic echo /camera1/imu/header/seq
rostopic echo /camera2/imu/header/seq
```
I see no interference. Each topic is sequential, increments of 1 between messages, and each starts at a different time (the time of the matching `rostopic` subscription)

akirayou, 2019-05-22:
Sorry,I misunderstood on  local static variable.(I didn't understand on class_loader) 
Please just forget it.

doronhi, 2019-05-22:
Done. Thanks.

closed 2019-05-22 by doronhi

akirayou, 2019-05-22:
Thank you! But I found  this now. hmm...  I'll try PR some day just for refactoring.
http://wiki.ros.org/CppStyleGuide#Static_class_variables