realsenseai/realsense-ros #777: Disable tf publishing in realsense2_camera node

martinakos, 2019-05-20:
When simulating in gazebo a robot that uses the D415 I use the _d415.urdf.xacro description. This description creates a static tf tree for the cameras in the D415. However, in the real robot using a D415 camera the tf tree is generated in the realsense2_camera node and the tree is different to the one described in _d415.urdf.xacro. I would like to disable this tf generation and just tell the realsense2_camera node which frames from the loaded _d415.urdf.xacro description the published messages should refer to. 
In this old issue [#9](https://github.com/IntelRealSense/realsense-ros/issues/9) it talks about the publish_tf parameter and it seems to be related to what I'm trying to achieve. However, that's a very old post and there is not trace of publish_tf in the code anymore. 
Is there any way to achieve what I want?

BriceRenaudeau, 2019-05-20:
You can just comment the include in the xacro : 
``
The mesh is still display using the package path : 
```
<geometry>
        <mesh filename="package://realsense2_camera/meshes/d435.dae" />
</geometry>
```

martinakos, 2019-05-20:
I actually would like to do the oposite, disable the tf generation in the realsense2_camera node and preserve the tf tree defined in the urdf.xacro. Just maybe disable loading the gazebo plugin when I'm using the real robot.

JessyXie, 2019-06-26:
Hi, does publish_tf flag still exist in the latest version?

doronhi, 2019-12-09:
Can be closed with #902 ?

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@JessyXie Could you please check if https://github.com/IntelRealSense/realsense-ros/pull/902 meet your need? And any other questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@JessyXie  Sorry that we didn't get response from you for weeks. Will close this issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport