realsenseai/realsense-ros #779: Disable Relocalization/Loop Closure for T265

manomitbal, 2019-05-20:
Is there a way to disable Relocalization/Loop closure on the T265 using a param or code change until the Relocalization API is considered stable? I am planning to use odometry output as a differential instead, Would that be possible?

BriceRenaudeau, 2019-05-20:
I think it is better to directly use the twist from the odom/sample topic (that's what I am doing). 
They are not differential hence they don't present the jumps from the pose. 
And the covariances are coherent.

manomitbal, 2019-05-20:
That makes sense, thanks @BriceRenaudeau !

macmason, 2019-06-19:
Upvote here; we want to use this sensor as a "pure" odometer inside a much larger SLAM system, but loop closures make that impossible.

doronhi, 2019-06-26:
It sounds like a valid use case and we'll take it into consideration in inner discussions. @radfordi

radfordi, 2019-06-27:
Thanks for the feedback @manomitbal and @macmason.  We have a few different layers we can disable.  I see about exposing them.

RealSenseCustomerSupport, 2019-10-09:
----------------------------------------------

Hello,

There is now a RS2_OPTION_ENABLE_RELOCALIZATION flag in the 'rs2_options' enum. *Omitting* this flag should accomplish what you're looking for.

You'll also find a control_sensor_options() method in librealsense/examples/sensor-control/rs-sensor-control.cpp that demonstrates how to change sensor options.

This should be sufficient for your needs. Is there anything else I can assist you with this matter?

Thanks

nacorti, 2019-10-13:
After looking at this topic, [this](https://github.com/introlab/rtabmap/issues/409) issue from RTabMap, and [this](https://github.com/IntelRealSense/realsense-ros/issues/936) question from the realsense-ros issue tracker, is there any plan to publish relocalization events to output? 

I'd prefer to use the T265's strengths and not rely on an app like rtabmap to perform relocalization - OTOH, T265 loop closure conflicts with rtabmap's, and there's no way to properly optimize the map with both localization processes running.

If not, is there a best practice to detect relocalization events from looking at the other output streams?

radfordi, 2019-10-15:
We have added a few options https://github.com/IntelRealSense/librealsense/pull/4321 to disable different parts of the internal mapping on the T265.   I would suggest leaving mapping enabled, but disabling relocalization and jumping.  This may avoid the conflict with RTabMap while still giving good results.  We are curious to hear which combinations work well for you.

With respect to relocalization events, we have just enabled them, https://github.com/IntelRealSense/librealsense/pull/4936.  Please let us know your thoughts on the API.

doronhi, 2019-10-15:
All the options add in [IntelRealSense/librealsense#4321](https://github.com/IntelRealSense/librealsense/pull/4321) are added automatically in realsense2_camera. As for the relocalization events, we still need to integrate those. Any thoughts about the ROS API for that. Is there a convention or a better way then publishing another topic? 
The issue is also referenced [here](https://github.com/IntelRealSense/realsense-ros/issues/936)

cimbar, 2019-11-14:
> Hello,
>
> There is now a RS2_OPTION_ENABLE_RELOCALIZATION flag in the 'rs2_options' enum. _Omitting_ this flag should accomplish what you're looking for.
>
> You'll also find a control_sensor_options() [...]

how can we disable RS2_OPTION_ENABLE_RELOCALIZATION flag?

doronhi, 2019-11-17:
You could use rqt_reconfigure to turn "enable_relocalization" off or you could add the following lines to your launch file:
```
  <rosparam>
      /camera/tracking_module/enable_relocalization: false
  </rosparam>
```

RealSenseCustomerSupport, 2019-11-18:
----------------------------------------------

Thanks doronhi.

cimbar, manomitbal...will you be needing further assistance with this issue?

Thanks

cimbar, 2019-11-19:
It solves my problem on ros side, thanks. Additionally, it would be better to explain how rs2_option flags can be disabled from librealsense

RealSenseCustomerSupport, 2019-11-19:
----------------------------------------------

If you look at the _change_sensor_options()_ method in Intel RealSense SDK 2.0\samples\sensor-control\api_how_to.h you will find an example. You should be able to simple set RS2_OPTION_ENABLE_RELOCALIZATION to 0 to accomplish what you want.

closed 2019-11-27 by RealSenseCustomerSupport