realsenseai/realsense-ros #780: Install librealsense-dkms and -dev from rosdep

veimox, 2019-05-21:
At Blue Ocean Robotics we use rosdep to install `librealsense-dkms` and `librealsense-dev`. I think you might also want to use this method too instead of making it manual. Instead of doing the PR in rosdistro i think you should take ownership of this. You are more than welcome to either copy-paste our script or directly point to it.

The steps to be done is to create an entry in `base.yaml` in [rosdistro](https://github.com/ros-infrastructure/rosdistro) with the content:

```yaml
librealsense2:
  ubuntu:
    xenial:
      source:
        uri: "https://bitbucket.org/api/2.0/snippets/blue-ocean-robotics/aendn6/master/files/librealsense2.rdmanifest"
```

We only target Ubuntu Xenial, but you need to adapt the entry for your use case. Then, when the entry `librealsense2` is found in the package.xml, that script will be run allowing thus to install the libraries. 

As a note, in the script of the uri there *must* exist this uri that points to a tar file. This can be empty as the one used (that is actually provided by magazino)

This is issue is aligned with #704 and is a pre-step before blooming because AFAIK it is a requirement that a ros package installs its own dependencies automatically.

doronhi, 2019-05-21:
Thanks for the info!

doronhi, 2019-06-02:
@veimox ,
Started following your suggestion. Proposed [PR21430](https://github.com/ros/rosdistro/pull/21430)

closed 2019-11-11 by RealSenseCustomerSupport