realsenseai/realsense-ros #783: Given depth profile is not supported by current device!--When trying to change the resolution of D415

mymrblack, 2019-05-23:
Hello, I was trying to change the resolution of D415 on ROS platform. Then I met a problem:

If I just use {roslaunch realsense2_camera rs_camera.launch camera:=cam_1 serial_no:=839512060214 filters:=spatial,temporal,pointcloud}, then I can successfully run the nodes.(Though it will output a warn like: [ WARN] [1558599157.236867531]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Format: Y8, Width: 640, Height: 480, FPS: 30)

But if I change the command to { roslaunch realsense2_camera rs_camera.launch camera:=cam_1 serial_no:=839512060214 filters:=spatial,temporal,pointcloud depth_height:=360 color_height:=360 fisheye_height:=360 infra1_height:=360 infra2_height:=360}, then I can not run it. The output in the command window was like what the SUMMARY shows. This command works successfully on another camera, so I am wandering what's the reason of these errors.

BTW, the lsusb results are:
Bus 002 Device 006: ID 8086:0ad3 Intel Corp. 
Bus 002 Device 003: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 002 Device 002: ID 0bda:0401 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 014: ID 8086:0ad3 Intel Corp. 
Bus 001 Device 007: ID 046d:c534 Logitech, Inc. Unifying Receiver
Bus 001 Device 006: ID 060b:7a03 Solid Year 
Bus 001 Device 004: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 003: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Bus 001 Device 002: ID 0bda:5401 Realtek Semiconductor Corp. RTL 8153 USB 3.0 hub with gigabit ethernet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

SUMMARY
========

PARAMETERS
 * /cam_1/realsense2_camera/accel_fps: 1000
 * /cam_1/realsense2_camera/accel_optical_frame_id: cam_1_accel_optic...
 * /cam_1/realsense2_camera/align_depth: False
 * /cam_1/realsense2_camera/aligned_depth_to_color_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_fisheye_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_infra1_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/aligned_depth_to_infra2_frame_id: cam_1_aligned_dep...
 * /cam_1/realsense2_camera/base_frame_id: cam_1_link
 * /cam_1/realsense2_camera/color_fps: 30
 * /cam_1/realsense2_camera/color_frame_id: cam_1_color_frame
 * /cam_1/realsense2_camera/color_height: 360
 * /cam_1/realsense2_camera/color_optical_frame_id: cam_1_color_optic...
 * /cam_1/realsense2_camera/color_width: 640
 * /cam_1/realsense2_camera/depth_fps: 30
 * /cam_1/realsense2_camera/depth_frame_id: cam_1_depth_frame
 * /cam_1/realsense2_camera/depth_height: 360
 * /cam_1/realsense2_camera/depth_optical_frame_id: cam_1_depth_optic...
 * /cam_1/realsense2_camera/depth_width: 640
 * /cam_1/realsense2_camera/enable_color: True
 * /cam_1/realsense2_camera/enable_depth: True
 * /cam_1/realsense2_camera/enable_fisheye: True
 * /cam_1/realsense2_camera/enable_imu: True
 * /cam_1/realsense2_camera/enable_infra1: True
 * /cam_1/realsense2_camera/enable_infra2: True
 * /cam_1/realsense2_camera/enable_pointcloud: False
 * /cam_1/realsense2_camera/enable_sync: False
 * /cam_1/realsense2_camera/filters: spatial,temporal,...
 * /cam_1/realsense2_camera/fisheye_fps: 30
 * /cam_1/realsense2_camera/fisheye_height: 360
 * /cam_1/realsense2_camera/fisheye_optical_frame_id: cam_1_fisheye_opt...
 * /cam_1/realsense2_camera/fisheye_width: 640
 * /cam_1/realsense2_camera/gyro_fps: 1000
 * /cam_1/realsense2_camera/gyro_optical_frame_id: cam_1_gyro_optica...
 * /cam_1/realsense2_camera/infra1_fps: 30
 * /cam_1/realsense2_camera/infra1_frame_id: cam_1_infra1_frame
 * /cam_1/realsense2_camera/infra1_height: 360
 * /cam_1/realsense2_camera/infra1_optical_frame_id: cam_1_infra1_opti...
 * /cam_1/realsense2_camera/infra1_width: 640
 * /cam_1/realsense2_camera/infra2_fps: 30
 * /cam_1/realsense2_camera/infra2_frame_id: cam_1_infra2_frame
 * /cam_1/realsense2_camera/infra2_height: 360
 * /cam_1/realsense2_camera/infra2_optical_frame_id: cam_1_infra2_opti...
 * /cam_1/realsense2_camera/infra2_width: 640
 * /cam_1/realsense2_camera/json_file_path: 
 * /cam_1/realsense2_camera/pointcloud_texture_index: 0
 * /cam_1/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /cam_1/realsense2_camera/rosbag_filename: 
 * /cam_1/realsense2_camera/serial_no: 839512060214
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /cam_1/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[cam_1/realsense2_camera_manager-1]: started with pid [30033]
process[cam_1/realsense2_camera-2]: started with pid [30034]
[ INFO] [1558599108.896374841]: Initializing nodelet with 6 worker threads.
[ INFO] [1558599108.952916374]: RealSense ROS v2.1.2
[ INFO] [1558599108.953019382]: Running with LibRealSense v2.17.0
[ INFO] [1558599109.308124745]: getParameters...
[ INFO] [1558599109.547341122]: setupDevice...
[ INFO] [1558599109.547457026]: JSON file is not provided
[ INFO] [1558599109.547857058]: ROS Node Namespace: cam_1
[ INFO] [1558599109.548055682]: Device Name: Intel RealSense D415
[ INFO] [1558599109.548140994]: Device Serial No: 839512060214
[ INFO] [1558599109.548216642]: Device FW version: 05.09.13.00
[ INFO] [1558599109.548294466]: Device Product ID: 0x0AD3
[ INFO] [1558599109.548368354]: Enable PointCloud: On
[ INFO] [1558599109.548442082]: Align Depth: Off
[ INFO] [1558599109.548510210]: Sync Mode: On
[ INFO] [1558599109.548646818]: Device Sensors: 
[ INFO] [1558599109.548767267]: Stereo Module was found.
[ INFO] [1558599109.548941507]: RGB Camera was found.
[ INFO] [1558599109.549185539]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1558599109.549285603]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1558599109.549352099]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1558599109.549447907]: setupPublishers...
[ INFO] [1558599109.572709839]: Expected frequency for depth = 30.00000
[ INFO] [1558599109.722284698]: Expected frequency for infra1 = 30.00000
[ INFO] [1558599109.865404194]: Expected frequency for infra2 = 30.00000
[ INFO] [1558599110.006975690]: Expected frequency for color = 30.00000
[ INFO] [1558599110.159750103]: setupStreams...
[ WARN] [1558599110.162639097]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Format: Z16, Width: 640, Height: 360, FPS: 30
[ WARN] [1558599110.162942841]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 1, Format: Y8, Width: 640, Height: 360, FPS: 30
[ WARN] [1558599110.163155641]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Format: Y8, Width: 640, Height: 360, FPS: 30
[ WARN] [1558599110.164713050]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Format: RGB8, Width: 640, Height: 360, FPS: 30
[ INFO] [1558599110.164977178]: Add Filter: spatial
[ INFO] [1558599110.166916603]: Add Filter: temporal
[ INFO] [1558599110.168817692]: Add Filter: pointcloud
[ INFO] [1558599110.170819229]: num_filters: 3
[ INFO] [1558599110.170952669]: publishStaticTransforms...
[ERROR] [1558599110.171146429]: Given depth profile is not supported by current device!
[FATAL] [1558599110.560775011]: Failed to load nodelet '/cam_1/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[cam_1/realsense2_camera_manager-1] process has died [pid 30033, exit code 1, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/tianbot/.ros/log/689f35a0-7d28-11e9-8be7-00044bc5333b/cam_1-realsense2_camera_manager-1.log].
log file: /home/tianbot/.ros/log/689f35a0-7d28-11e9-8be7-00044bc5333b/cam_1-realsense2_camera_manager-1*.log
[cam_1/realsense2_camera-2] process has died [pid 30034, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/tianbot/.ros/log/689f35a0-7d28-11e9-8be7-00044bc5333b/cam_1-realsense2_camera-2.log].
log file: /home/tianbot/.ros/log/689f35a0-7d28-11e9-8be7-00044bc5333b/cam_1-realsense2_camera-2*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done_

doronhi, 2019-06-04:
regarding your first run command, I think you should turn Advanced-Mode On. To do so run realsense-viewer and use the button "more".
Regarding your second command, run `rs-enumerate-devices`. Do you see the desired profiles in the list of supported profiles?

doronhi, 2019-06-10:
Did you manage any progress? Any additional data?

closed 2019-06-11 by mymrblack