realsenseai/realsense-ros #787: Publish Raw image to ROS

CaoKha, 2019-05-27:
Hi,
I want to publish the data from:
```
rs2::frame depth_frame = frames.get_depth_frame();
rs2::frame color_frame = frames.get_color_frame();
auto depth_timestamp = depth_frame.get_timestamp();
auto color_timestamp = color_frame.get_timestamp();
```
to ROS and save it as ros bag. I don't want to use the function in example record_playback because I want to do something with the frames in real time (such as save them to a file frame by frame and matching them with data from IMU)
But I don't know how. I am new to code btw. Thanks in advance.

doronhi, 2019-06-10:
Sorry for the long delay.
It sounds to me like you don't want to subscribe to the topics published by the wrapper but to connect directly to the device, read the images, do something and then publish them as ROS topics. Am I correct.
That's basically what the realsense2_wrapper does. So looking at the source code would be a good start.
The part where the images are available before being published is in the function `BaseRealSenseNode::frame_callback`.

CaoKha, 2019-06-18:
Hi @doronhi,
Can I ask why in function publishframe, you put an object rs2::frame but not a reference?

doronhi, 2020-01-30:
rs2::frame holds a pointer to an inner librealsense frame objects that holds a reference counter. So we are not copying data by giving the function an object but just increasing the reference counter by 1.

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@CaoKha Any other questions about this ticket?

closed 2020-02-11 by CaoKha

wangcongrobot, 2021-03-15:
Hi, can I use python to get the raw images?