realsenseai/realsense-ros #788: Depth and Pointcloud from T265

mzahana, 2019-05-28:
It would be nice if the following can be wrapped into a ROS node to provide depth and pointcloud from T265 as ROS messages.
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py

Related issue:
https://github.com/IntelRealSense/librealsense/issues/3964

RealSenseCustomerSupport, 2019-08-26:
----------------------------------------------

Hi @mzahana

Thanks for the feedback.

mzahana, 2019-08-26:
You are welcome.

closed 2019-10-02 by RealSenseCustomerSupport

rjam92, 2020-10-15:
Hello, was this accomplished?

rjam92, 2020-10-16:
Hello, 
Was this accomplished?

mzahana, 2020-10-16:
@rjam92 I didn't continue working on this for quite sometime. I am not sure what the state is. I believe, support will likely not be maintained for T265.

whyscience, 2025-03-25:
https://github.com/IntelRealSense/librealsense/blob/v2.53.1/wrappers/python/examples/t265_stereo.py