realsenseai/realsense-ros #790: melodic build broken

peci1, 2019-05-29:
@doronhi @v-lopez Guys, commits bf9b16fe45eb60d2d752b76e717615b9fd965472 and cfd5d0c9c40f24fbdb24f7c69ad604031b58605c just broke the build on melodic.

Please, release ddynamic_reconfigure to melodic. Thanks. I know I can build it from source, but this one I think can easily work on melodic and should be also accessible from binary repos.

v-lopez, 2019-05-29:
Requested the release here:
https://github.com/ros/rosdistro/pull/21380

pedropgusmao, 2019-06-04:
Please, how should we install this?

v-lopez, 2019-06-04:
In the meantime you should be able to compile ddynamic_reconfigure yourselves, cloning it from: https://github.com/pal-robotics/ddynamic_reconfigure

The kinetic-devel branch should be ok

constructive-realities, 2019-06-07:
should we just build this an install it from the user's home, or from within the ros workspace src folder?

constructive-realities, 2019-06-07:
Confirming valid workaround to add ddynamic_reconfigure inside the workspace, not in the users HOME (FYI)

doronhi, 2019-06-10:
As the workaround solution is confirmed and we wait for the release of ddynamic_reqonfigure for melodic, I think this issue can be closed.

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Hi @peci1,

Did you have a chance to try the workaround solution? Please let us know if you need any
further help with this matter.

Thank you!

peci1, 2019-10-30:
Hi, since ddynamic_reconfigure is now properly released, this issue can be closed. It is just needed to clear your catkin workspace when switching from in-repo ddynamic to the released one.

closed 2019-10-30 by peci1