realsenseai/realsense-ros #794: no rclcpp::QoS on ubuntu18.04 ROS2/crystal

nizqsut, 2019-05-31:
I try to use D435I in ROS2.
Realsense-viewer is ok. And ROS2 examples run ok.
When I build ros2_intel_realsense meet issue like this:
src/ros2_intel_realsense/realsense_ros2_camera/src/realsense_camera_node.cpp:1365:11: error: ‘QoS’ in namespace ‘rclcpp’ does not name a type
   rclcpp::QoS qos;

Trying to fix this issue.

doronhi, 2019-06-10:
realsense wrapper for ROS2 is maintained [here](https://github.com/intel/ros2_intel_realsense) 
You'd better post the question there.

nizqsut, 2019-06-11:
> realsense wrapper for ROS2 is maintained [here](https://github.com/intel/ros2_intel_realsense)
> You'd better post the question there.

I fixed the problem by building ROS2 from source code.

closed 2019-10-29 by RealSenseCustomerSupport