realsenseai/realsense-ros #796: Global Camera Time

pavloblindnology, 2019-05-31:
Hello.
The latest librealsense version introduced Global Camera Time frame (currently works for D400 and L500 only) https://github.com/IntelRealSense/librealsense/pull/3909
What are your plans to use this feature for ROS stamping?
This would address https://github.com/IntelRealSense/realsense-ros/issues/419 issue.

doronhi, 2019-06-02:
Global Time Stamp is On by default so it will be used and correct the time drifting between the camera and the host computer.
The mechanism inside the ROS wrapper is now used to synchronize times to the ROS time which represents time that could be kept by a different machine.
I expect that since ROS times are usually based on a time-stable machines, not much drifting will occur now but the idea for the future is to copy the time-correction mechanism of Global Time Stamp to sync between computer time and ROS time as well.
I agree that syncing once at initialization is not enough.

pavloblindnology, 2019-06-03:
Well, actually, Global Time Stamping On doesn't change the main realsense-ros behavior - i.e., the 1st frame arrived is prescribed 0 latency. Which will keep to be ~0 with time, thanks to Global Time Stamping On. This means you cannot do any latency measurements, since they are 0.
Concerning the sync between computer time and ROS time - you don't need to do this. ROS time IS computer time unless we're in simulation (http://wiki.ros.org/Clock). On multiple machines config you need to sync computers' wall clocks instead, using chrony or whatever.
So, seems we just need to use frame's global timestamp (whenever available) straight as a ROS timestamp, without any modifications.

doronhi, 2019-06-04:
Time correction inside the wrapper is still needed for devices that do not support Global Time Stamp and for cases where metadata is not available.
So, is this issue about the wrapper's time correction being redundant, at least when global time stamp is available, or do you find it causes errors?

pavloblindnology, 2019-06-04:
Yes, it's redundant when global time stamp is available. For other cases (metadata is not available) you have nothing to do but to use this option, though it's quite rough.

doronhi, 2019-06-05:
I Agree. Thanks.

RealSenseCustomerSupport, 2019-12-12:
----------------------------------------------

@pavloblindnology Any other questions about this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@pavloblindnology Could you please update? Thanks!

pavloblindnology, 2020-01-27:
Well. The issue is still there. With current implementation, the timestamp domain of the device is totally neglected, and frames' timestamps are converted to ROS times by initial frame synchronization (which is assumed to have latency 0).

doronhi, 2020-07-27:
The correction to ROS time when global time stamp is available is indeed redundant, but it's also negligible. Instead of having an "if" section that chooses different behavior for "global time stamp" and other time stamps, the current implementation is generic and the time stamp domain is not important.

RealSenseSupport, 2020-08-13:
@pavloblindnology  With doronhi's clarification, do you have any other questions about this issue? Looking forward to your update. Thanks!

RealSenseSupport, 2020-08-21:
@pavloblindnology Could you please update if still have questions about this? Thanks!

RealSenseSupport, 2020-09-08:
@pavloblindnology Sorry that we don't get response from you for weeks. Will close this accordingly. Please feel free to create another one if you still have issues or questions. Thanks!

closed 2020-09-08 by RealSenseSupport

hichamhendy, 2022-02-03:
Does the reslasense sdk for L515 have the capability to switch between internal camera clock and computer clock to stamp the messages?
I am trying to use to an external IMU on lieu of internals's? as described here https://github.com/introlab/rtabmap/issues/821#issue-1120719786

However, that works just fine with D435 yet not L515.

Does anybody have an idea what should I change/switch in the ROS wrapper to allow that?