realsenseai/realsense-ros #797: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is.

Achllle, 2019-06-01:
using realsense D415 on librealsense 2.22, FW 5.11.4, and realsense_ros v2.2.4. Tested with two different realsenses.

When I run `roslaunch realsense2_camera rs_camera.launch`, I continuously get the following: 
```
cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
```

When I launch realsense-viewer everything seems to be as expected and I can view all the streams in RViz.

doronhi, 2019-06-05:
Could you elaborate more? what other nodes are running on your system?
Could it be related to #315 ?

Achllle, 2019-06-11:
I'm afraid that's all there is.

I tried on a different machine too. I also tried downgrading to librealsense 2.18.0, realsense_ros v1.1.4, FW 5.10.13.
No other nodes running. My workaround is using logging's filters to get rid of that specific message but that doesn't seem right. I'll post the launch output asap.

doronhi, 2019-06-12:
I also don't like masking the errors as a principle but I can't reproduce.
Hopefully some new data will be available soon.

Achllle, 2019-06-14:
I tested the launch for a D435, getting the same error. Here's the output:
```
$ roslaunch realsense2_camera rs_camera.launch 
[omitted]

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
[... 44 lines cut ...]
[ INFO] [1560529631.474184067]: Expected frequency for depth = 30.00000
[ INFO] [1560529631.538741091]: Expected frequency for infra1 = 30.00000
[ INFO] [1560529631.598107616]: Expected frequency for infra2 = 30.00000
[ INFO] [1560529631.656474959]: Expected frequency for color = 30.00000
[ INFO] [1560529631.719630900]: setupStreams...
 14/06 09:27:11,720 WARNING [140074139338496] (sensor.cpp:367) Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]
[ INFO] [1560529631.722660706]: depth stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1560529631.723210776]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1560529631.723685502]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1560529631.727753092]: color stream is enabled - width: 640, height: 480, fps: 30
[ INFO] [1560529631.729685795]: insert Depth to Stereo Module
[ INFO] [1560529631.730020658]: insert Color to RGB Camera
[ INFO] [1560529631.730134158]: insert Infrared to Stereo Module
[ INFO] [1560529631.730213969]: insert Infrared to Stereo Module
[ INFO] [1560529631.744590697]: num_filters: 0
[ INFO] [1560529631.744750044]: SELECTED BASE:Depth, 0
[ INFO] [1560529631.747460978]: RealSense Node Is Up!
[ INFO] [1560529631.747534201]: Setting Dynamic reconfig parameters.
[ INFO] [1560529631.971799748]: Done Setting Dynamic reconfig parameters.
[ERROR] [1560529632.671337906]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1560529632.703835509]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1560529632.737101471]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1560529632.770510191]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1560529632.803846516]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
```

doronhi, 2019-06-16:
Still no idea..
Do you run some kind of inquiry? like `rostopic hz ...` or something else that subscribes on the images?
What puzzles me is that the the only reference to cv_bridge I find in the realsense2_wrapper code is in the publishFrame function and it is active only when images are being subscribed to which is not your case as you mentioned that no other node is running.
What OS do you run on? What are the system characteristics in term of memory and CPU?

Achllle, 2019-06-24:
You're right. We had a hidden `systemd` service running in the background that listens to all topics for logging and monitoring purposes. Stopping that made the error disappear.

closed 2019-06-24 by Achllle

danielTobon43, 2019-11-22:
@Achllle Can you post the solution? Thank you

Achllle, 2019-11-25:
As written above in my last comment, I resolved the issue by stopping all listeners that were subscribing to the image topic. In our case this was a monitoring service running in the background. Run `rostopic info` on the camera topic and it'll give you a list of subscribers.
Your listener probably is doing some conversion wrong, but I don't know exactly what.

hichamhendy, 2022-01-26:
In this regard, my listener is rosbag which I do need, which actually subscribes all available topics, Is there any way to avoid/solve this?

Achllle, 2022-01-26:
The listener is not using the right conversion. If that indeed is rosbag, it might be a bug within rosbag itself.

FPSychotic, 2023-05-26:
I had the same issue, upgraded same issue, installed in two different ways , same issue. Any clue, it is an bad bug as don't allow use bags in my case

pavan-0328, 2023-06-09:
i am getting the issue when is do rosbag record

MartyG-RealSense, 2023-06-09:
Hi @pavan-0328  Would it be possible for you to perform rosbag record without launching the RealSense ROS wrapper in order to confirm whether the problem is in ROS and not with the RealSense wrapper?  You can record all currently published topics with the instruction **rosbag record -a**

pavan-0328, 2023-06-09:
i used **rosbag record -a** only. and i tested what you have suggested and the problem is with Realsense Ros wrappers.

monajalal, 2023-11-15:
@doronhi I have the same issue with Intel RealSense D435 camera and ros noetic inside a docker.

https://robotics.stackexchange.com/questions/105438/depth-aligned-capture-doesnt-yield-same-number-of-rgb-and-depth-frames-cv-br