realsenseai/realsense-ros #798: easy way for using D435 upside down

TimoVFX, 2019-06-02:
Hi,

we are building an autonomous robot using two intel d435 in our perception pipeline. Due to the cable management we have to mount one camera upside down. 

Is there a **easy way** within this ROS wrapper to rotate the image by 180°? 

I tried to use the image_rotate node of ROS without any success. (nevertheless it is not recommended for further processing) If the only way is to write a node and rotated it via OpenCV or similar we will do this, i was just wondering if i am missing something obvious. 

Thanks,
Timo

MartyG-RealSense, 2019-06-03:
Are you mounting one camera upside down so that one USB cable is on the left side and the other is on the right side please, so that the two cables do not come into contact with each other?

I wonder if an easier solution than flipping the image that you could consider first would be to use a USB-C cable with a right-angled "L" connector with one of the cameras so that the L-connector cable comes in from the back instead of sidewards.

[image]

TimoVFX, 2019-06-03:
> Are you mounting one camera upside down so that one USB cable is on the left side and the other is on the right side please, so that the two cables do not come into contact with each other?

Yes that was the intend of the design, unfortunately it is not possible to exchange the cable, they are specially shielded cable against electro magnetic interference. Ordering new ones would take too long... Also they are mounted in a housing to be protected against rain. Maybe we can rethink the housing design. 

So, there is no easy way of rotating the image?

MartyG-RealSense, 2019-06-03:
I just suggested the L-connector cable first because it had the potential to be a much simpler solution than flipping the image.  Image-flipping is certainly possible though.

If you want to do it in ROS and not use its image_rotate function, and also would prefer to avoid doing it with OpenCV, maybe you could flip the **TF** in ROS instead of flipping the image.

https://answers.ros.org/question/242894/how-to-flip-a-camera-image-upside-down/

For further suggestions, you could google for 'flip image upside down camera ros'.

TimoVFX, 2019-06-03:
Thanks for your help Marty, I ended up flipping the image with OpenCV before publishing to the topic. Since i didnt got the TF flipping working.

closed 2019-06-03 by TimoVFX

msacco22, 2019-07-17:
@TimoVFX could you kindly elaborate where you did the flipping code. Is it in the base_realsense_node.cpp?