realsenseai/realsense-ros #800: Initial hardware reset not working on t265

ineslara21, 2019-06-04:
Hi!
I was trying to perform initial hardware reset using the tracking camera t265 but I only get an exception: `An exception has been thrown: hardware_reset is not implemented for this device!`. 

I'm using SDK version 2.22 built from source from https://github.com/IntelRealSense/librealsense 

I would like to know how to use `hardware_reset()` function or if you have any ideas about how could I implement this hardware reset. 

Thanks in advance!

MartyG-RealSense, 2019-06-04:
Hi there!

An initial response to your question was provided 5 days ago.  the status of that case remains Open, meaning that the RealSense team members on this forum will come back to it if they have an update that they can provide.   Thanks!

https://github.com/IntelRealSense/librealsense/issues/4113

ineslara21, 2019-07-23:
The status of that case is now closed. 
The solution is on 
https://github.com/IntelRealSense/librealsense/issues/4113
Thanks

closed 2019-07-23 by ineslara21