realsenseai/realsense-ros #805: ddynamic_reconfigure edition 0.2.0 missing 'registerVariable'; Invoking "make -j8 -l8" failed

goliasjd, 2019-06-06:
I have made sure that all the software is up to date, but when I try to run
catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
I get the following error message:

/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)’:
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:281:22: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘registerVariable’
             ddynrec->registerVariable<bool>(
                      ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:281:39: error: expected primary-expression before ‘bool’
             ddynrec->registerVariable<bool>(
                                       ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:283:89: error: expected primary-expression before ‘,’ token
               [option, sensor](bool new_value) { sensor.set_option(option, new_value); },
                                                                                         ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:309:24: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘registerVariable’
               ddynrec->registerVariable<int>(
                        ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:309:41: error: expected primary-expression before ‘int’
               ddynrec->registerVariable<int>(
                                         ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:311:92: error: expected primary-expression before ‘,’ token
                   [option, sensor](int new_value) { sensor.set_option(option, new_value); },
                                                                                            ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:329:28: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘registerVariable’
                   ddynrec->registerVariable<double>(
                            ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:329:45: error: expected primary-expression before ‘double’
                   ddynrec->registerVariable<double>(
                                             ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:331:99: error: expected primary-expression before ‘,’ token
                       [option, sensor](double new_value) { sensor.set_option(option, new_value); },
                                                                                                   ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:357:22: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘registerEnumVariable’
             ddynrec->registerEnumVariable<int>(
                      ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:357:43: error: expected primary-expression before ‘int’
             ddynrec->registerEnumVariable<int>(
                                           ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:359:90: error: expected primary-expression before ‘,’ token
                 [option, sensor](int new_value) { sensor.set_option(option, new_value); },
                                                                                          ^
/home/joel/catkin_ws/src/realsense2_camera/src/base_realsense_node.cpp:363:14: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘publishServicesTopics’
     ddynrec->publishServicesTopics();
              ^
realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:1500: recipe for target 'realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

All previous responses to similar problems have not worked.
Is there a new solution to this issue?

mysablehats, 2019-06-09:
Me too post. Tried adding ddynamic_reconfigure repo from awesomebytes and this happened. looks like linking/include error. without the ddynamic_reconfigure package it fails earlier with a:
```

CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "ddynamic_reconfigure" with any of the following names:

    ddynamic_reconfigureConfig.cmake
    ddynamic_reconfigure-config.cmake

  Add the installation prefix of "ddynamic_reconfigure" to CMAKE_PREFIX_PATH
  or set "ddynamic_reconfigure_DIR" to a directory containing one of the
  above files.  If "ddynamic_reconfigure" provides a separate development
  package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  realsense-ros/realsense2_camera/CMakeLists.txt:8 (find_package)

```

doronhi, 2019-06-10:
Are you using ROS kinetic? if not then check #790 
This error: 
`class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘registerVariable’
ddynrec->registerVariable(` 
suggest that version 0.2.0 of ddynamic_reconfigure is not installed. You mentioned that all the software is up to date but that's why I'll ask you to check again that the right version of ddynamic_reconfigure (0.2.0) is installed (`apt list | grep ddynamic`)?
If you want to install ddynamic_reconfigure from source, the latest version of ddynamic_reconfigure is not maintained by awesomebytes but [here](https://github.com/pal-robotics/ddynamic_reconfigure).

goliasjd, 2019-06-10:
`ros-kinetic-ddynamic-reconfigure/xenial,now 0.2.0-0xenial-20190325-154841-0800 amd64 [installed]`

is returned when I run: `apt list | grep ddynamic`
so yes, I'm using kinetic and my version is current (I think), but for some reason it's still throwing the error. Any other ideas?

doronhi, 2019-06-11:
According to the log you post the function `registerVariable` is not available.
What header file is the compiler using?
To check, add the following line to the CMakeLists.txt file, at line 23, just after the find_package section:
`
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -H")
`
Then  run:
`catkin_make -DCMAKE_BUILD_TYPE=Release 2>&1 > /dev/null | grep ddynamic_reconfigure.h`
I get:
`
.. /opt/ros/kinetic/include/ddynamic_reconfigure/ddynamic_reconfigure.h
.... /opt/ros/kinetic/include/ddynamic_reconfigure/ddynamic_reconfigure.h
........ /opt/ros/kinetic/include/ddynamic_reconfigure/ddynamic_reconfigure.h
`
and when I check inside:
`cat /opt/ros/kinetic/include/ddynamic_reconfigure/ddynamic_reconfigure.h | grep registerVariable`
I get:
`
   * @brief registerVariable register a variable to be modified via the
  void registerVariable(const std::string &name, T *variable,
   * @brief registerVariable register a variable to be modified via the
  void registerVariable(const std::string &name, T current_value,
`

Do you see anything helpful here?

goliasjd, 2019-06-11:
Not sure why it wasn't working before, but the problem is solved. I just cleaned out and made a new catkin workspace, and running `catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release` no longer gives me any errors. Thanks for the help.

closed 2019-06-11 by goliasjd