realsenseai/realsense-ros #807: is this the correct way to get depth at (x,y) using cv_bridge?

npd26, 2019-06-07:
Hello,

can anyone guide me how to get z distance at particular point (x,y) when using realsense camera with cv_bridge in ROS?

```
`class ImageConverter
{
  ros::NodeHandle nh_;
  image_transport::ImageTransport it_;
  image_transport::Subscriber image_sub_;
  image_transport::Subscriber image_sub_1;
  image_transport::Publisher image_pub_;
  ros::Publisher _pub_pose;
  double distance = 0.0;
  double xr = 0.0;
  double yr = 0.0;

public:
  ImageConverter()
    : it_(nh_)
  {
    // Subscrive to input video feed and publish output video feed
    image_sub_ = it_.subscribe("/camera/color/image_raw", 1,
      &ImageConverter::imageCb, this);
    image_sub_1 = it_.subscribe("/camera/aligned_depth_to_color/image_raw", 1,
      &ImageConverter::imageCb1, this);
    image_pub_ = it_.advertise("/image_converter/output_video", 1);
    _pub_pose = nh_.advertise<geometry_msgs::PoseStamped>("object_pose", 1);
    cv::namedWindow(OPENCV_WINDOW);
  }

  ~ImageConverter()
  {
    cv::destroyWindow(OPENCV_WINDOW);
  }

  //getting depth
 
  void imageCb1(const sensor_msgs::ImageConstPtr& msg)
  {
    cv_bridge::CvImagePtr cv_ptr;
    try
    {
      cv_ptr = cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::TYPE_16UC1);
    }
    catch (cv_bridge::Exception& e)
    {
      ROS_ERROR("cv_bridge exception: %s", e.what());
      return;
    }
   
    **distance = 0.001*cv_ptr->image.at<u_int16_t>(xr, yr);**
    std::cout<<distance<<std::endl;
  }

//getting xr,yr

  void imageCb(const sensor_msgs::ImageConstPtr& msg)
  {
    cv_bridge::CvImagePtr cv_ptr;
    try
    {
      cv_ptr = cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::BGR8);
    }
    catch (cv_bridge::Exception& e)
    {
      ROS_ERROR("cv_bridge exception: %s", e.what());
      return;
    }
    
    // my try to detect circle
    cv::Mat src_gray,src_gray1;
    .....
    **here finding (xr,yr)**
    ....`
```

thanks

npd26, 2019-06-13:
any idea?

doronhi, 2019-06-13:
I have no working example that does that so, I would have to start by compiling your code and give it a try.
I guess that's what you already did and it didn't work. Is it a compilation problem?
What were the results? What is the error?

npd26, 2019-06-14:
@doronhi It is working. but when I run this ROS node, in starting It is taking wrong random depth and after several frames pass when OpenCV actually find (xr,yr) then It gives me correct depth. 
also, sometimes when OpenCV does not find (xr,yr), but then it gives me correct depth of plane but in reality there is no (xr,yr).

npd26, 2019-06-18:
@doronhi I found one python example, could you look at this?
_referenced issue above_

npd26, 2019-07-05:
@doronhi actually my goal  is similar like this function, rs2_deproject_pixel_to_point, **static void rs2_deproject_pixel_to_point(float point[3], const struct rs2_intrinsics * intrin, const float pixel[2], float depth)** but it is in realsense sdk, I want this with cv_bridge (OpenCV) in ROS. is there any idea?

eyalu, 2019-07-11:
I want to support @npd26 's request. My use case is similar to the "rs-measure" sdk example, yet with the current ros node functionality it can't be done. 

Or am I missing something?

doronhi, 2019-07-11:
There is a python script in the repository: [show_center_depth.py](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/scripts/show_center_depth.py)
Is this a good example of extracting distance using cv_bridge?

As for rs2_deproject_pixel_to_point, you have the pixel and depth (make sure you subscribe to the aligned_depth_to_color topic). The intrinsics are available in the "/camera/color/camera_info" topic and you can use the code [here](https://github.com/IntelRealSense/realsense-ros/blob/31fa75ec835650d319f2d5abd3f7d0ab66fd37f4/realsense2_camera/src/base_realsense_node.cpp#L1609) to see how they were converted from the originals that were used in the original rs2_deproject_pixel_to_point function.
I didn't test it myself though, so let me know what works for you.

npd26, 2019-07-11:
for rs2_deproject_pixel_to_point, we can use equation like this 2d-->3d, [https://ibb.co/y4jRpgR](url)
To get point(u,v) to (X,Y,Z) value, can we set R matrix as Identity matrix? what should be T matrix? is it [0,0,depth.at(u,v)]?

eyalu, 2019-07-13:
I'll start with a disclaimer that this is the first time I'm trying to do this kind of projection, so my apologies in advance if the horror it too great :-O

This is how I tried to implement the projection based on the references provided by @doronhi :
```
//Snippet of main function, gets x and y of a pixel to be measured in physical space:

	cv_bridge::CvImageConstPtr last_received_depth = cv_bridge::toCvCopy(*global_depth_image, sensor_msgs::image_encodings::TYPE_16UC1);
	float pixel_depth = last_received_depth->image.at<float>(y, x);
	cout << "Depth: " << pixel_depth << endl;
	rs2_intrinsics intr = global_cam_info;
	pixelTo3DPoint(x, y, pixel_depth, single_point);
	std::cout << "X: " << single_point.point.x << std::endl;
	std::cout << "Y: " << single_point.point.y << std::endl;
	std::cout << "Z: " << single_point.point.z << std::endl;
	std::cout << "-----------------------------------" << std::endl;

//Utility function
void pixelTo3DPoint(int x, int y, float depth, geometry_msgs::PointStamped &p){
	// X = (pixel_x - camera_intrinsics.ppx)/camera_intrinsics.fx *depth
	float real_x = (x - global_cam_info.K.at(2)) / global_cam_info.K.at(0) * depth;
	// Y (pixel_y - camera_intrinsics.ppy)/camera_intrinsics.fy *depth)
	float real_y = (y - global_cam_info.K.at(5)) / global_cam_info.K.at(4) * depth;
	p.point.x = real_x;
	p.point.y = real_y;
	p.point.z = depth;	
	return;
}
```

My readings are all over the place... any ideas as to what am I doing wrong?

eyalu, 2019-07-14:
Forgot to mention that the global depth image and the camera info globals are updated by subscribers to two topics:
/camera/aligned_depth_to_color/image_raw and /camera/aligned_depth_to_color/camera_info respectively

npd26, 2019-07-15:
only problem in this is, at starting in several frame it will be zero depth, after finding (u,v) it will work.

```
`class ImageConverter
{
  ros::NodeHandle nh_;
  image_transport::ImageTransport it_;
  image_transport::Subscriber image_sub_;
  image_transport::Subscriber image_sub_1;
  image_transport::Publisher image_pub_;
 
  
  double distance = 0.0;
  double x = 0.0;
  double y = 0.0;
 
public:
  ImageConverter()
    : it_(nh_)
  {
    // Subscrive to input video feed and publish output video feed
    image_sub_1 = it_.subscribe("/camera/aligned_depth_to_color/image_raw", 1,
      &ImageConverter::imageCb1, this);
    image_sub_ = it_.subscribe("/camera/color/image_raw", 1,
      &ImageConverter::imageCb, this);
    
    image_pub_ = it_.advertise("/image_converter/output_video", 1);

    cv::namedWindow(OPENCV_WINDOW);
  }

  ~ImageConverter()
  {
    cv::destroyWindow(OPENCV_WINDOW);
  }

  //getting depth
 
    void imageCb1(const sensor_msgs::ImageConstPtr& msg)
  {
    
    cv_bridge::CvImagePtr cv_ptr;
    try
    {
      cv_ptr = cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::TYPE_16UC1);
    }
    catch (cv_bridge::Exception& e)
    {
      ROS_ERROR("cv_bridge exception: %s", e.what());
      return;
    }
    
    this->distance = 0.001*cv_ptr->image.at<u_int16_t>(this->xr, this->yr);
    std::cout<<distance<<std::endl;
    
  }

//getting x,y

  void imageCb(const sensor_msgs::ImageConstPtr& msg)
  {
    cv_bridge::CvImagePtr cv_ptr;
    try
    {
      cv_ptr = cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::BGR8);
    }
    catch (cv_bridge::Exception& e)
    {
      ROS_ERROR("cv_bridge exception: %s", e.what());
      return;
    }
    
    // my try to detect circle
    cv::Mat src_gray,src_gray1;
    
    
    cv::cvtColor(cv_ptr->image, src_gray1, cv::COLOR_BGR2GRAY);

    //Here find (u,v),
    // then using above equation convert it to real (x,y)
    // also publish (x,y,z(distance))    
    
    cv::imshow(OPENCV_WINDOW, cv_ptr->image);
    cv::waitKey(3);

    // Output modified video stream
    image_pub_.publish(cv_ptr->toImageMsg());

  }
 
};`
```

eyalu, 2019-07-15:
Distance works fine, but x,y in physical space are still a problem :-(

npd26, 2019-07-16:
you have to get pixel (u,v) point and using above equation you can get (x,y,z) coordinates of that point in camera_color_optical_frame (which is located at RGB sensor center according to me but @doronhi can confirm that).

eyalu, 2019-09-10:
For sake of completeness, I finally managed to solve it, sorry it took me a while to post the answer.
You can take my previously posted code and change from image.at<float> to image.at<u_int16_t> as @npd26 suggested and that will work perfectly.
@npd26 note that your code should have x and y reversed in this line of code, due to that opencv accesses the image with row,column coordinates when using "at". Might solve you some annoying bad results :-)

In any case I think that since this workaround exists @doronhi you can close this, but if you happen to expose the pixelTo3dPoint as a service of the ros node, that would be greatly appreciated.

Thank you both!

ravijo, 2019-10-02:
Hi @eyalu and @npd26 

I made a ROS package to do the same. Please check it out [here](https://github.com/ravijo/ros_openpose).

Cheers!

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

Thanks @ravijo's contribution here.

closed 2019-12-05 by RealSenseCustomerSupport