realsenseai/realsense-ros #808: librealsense2 does not exist in rosdistro which causes error on `rosdep install`

furushchev, 2019-06-10:
The `librealsense2` package written in `package.xml` does not exist on rosdistro DB which causes an error when running `rosdep install`.
https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/package.xml#L28

```bash
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_camera: Cannot locate rosdep definition for [librealsense2]
```

Is it necessary to claim this package as a dependency in `package.xml`?

furushchev, 2019-06-10:
FYI, this behavior is not checked on the CI process of this repository.
I created a PR to include the building process to testing: https://github.com/IntelRealSense/realsense-ros/pull/806
The PR fails because of this error.

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

@furushchev Seems this issue resolve in latest ROS wrapper. Can we close this? Thanks!

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@furushchev  Could you please update? Thanks!

furushchev, 2019-12-20:
Sorry for being late!
I created a pull request to check if this issue is resolved.
https://github.com/IntelRealSense/realsense-ros/pull/1032
Once the testing ends successfully, I believe this issue can be closed.

tylerjw, 2019-12-22:
I'm trying to use this with `Melodic` but am running into a couple of issues related to this.  

Currently building `librealsense2` as a source package (through rosinstall) doesn't work because it's `package.xml` doesn't list all the required dependencies.

Secondly because it is only released as a kinetic package, installing it in the recommended way (debian install) doesn't work because then rosdep install fails with this error.

furushchev, 2019-12-23:
@RealSenseCustomerSupport It looks it is still not solved. Please look at #1032.

doronhi, 2019-12-23:
I am all for the idea behind #1032.
The CI run of #1032, did not fail because of the librealsense2 installation. That is because lines 10-11 in .travis.yaml which are responsible for it are still there.
It failed because of another test which for reasons I am yet unaware of is unstable in the Travis runs. I believe if you re-run the test it will pass.
[PR#23276](https://github.com/ros/rosdistro/pull/23276) for inclusion of librealsense v2.31.0 was merged and hopefully, once the next release of rosdistro is available, the ros installation of realsense2_camera and librealsense2 will work on Kinetic.
I am aware of issues regarding Melodic and it is in the pipeline.

furushchev, 2019-12-23:
@doronhi Thank you for the comment! :+1:  Please, then re-run the test when you think everything is ok.

doronhi, 2019-12-23:
I couldn't find the "rerun" button. I thought maybe it's unavailable for me because I'm am not the author of this PR.

furushchev, 2019-12-23:
@doronhi OK, I re-opened the pull request, which invokes test again.

RealSenseCustomerSupport, 2020-03-06:
----------------------------------------------

@furushchev PR#1032 is not merged yet. Do you want to keep this ticket open or can we close this and monitor PR status? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@furushchev What's your opinion? Could you please update? Thanks!

furushchev, 2020-03-17:
@RealSenseCustomerSupport Sorry for the late reply.
I confirmed that `librealsense2` library can now be installed by apt package `ros-melodic-librealsense2`, for instance in melodic.
However, the library installed via `apt-get` is too old and I cannot build this package with it.
To make matters worse (at least for me), it prevents to link librealsense2 built and installed from source since the library via `apt-get` is installed in `/opt/ros/melodic` which is declared in `PATH` and used by compiler more preferentially than `/usr/local` or `/usr` in vanilla Linux/ROS environment.

`rosdep` no more claims error during `rosdep install`, but I think it still need to let this issue be open unless the issue above is solved.

RealSenseCustomerSupport, 2020-03-20:
----------------------------------------------

@furushchev Ok. Thanks for your feedback! Let's keep it open till the issue resolved. Thanks!

doronhi, 2020-03-20:
librealsense2 v2.33.1, the latest version, is now available as ros-melodic-librealsense2.
realsense2-camera 2.2.13 is also available.

The packages were submitted for kinetic at the same time but updates for kinetic are still to be published.

RealSenseCustomerSupport, 2020-04-21:
----------------------------------------------

@doronhi Could you please tell if the updates for kinetic published or not? Thanks!

RealSenseSupport, 2020-07-02:
@furushchev New ROS wrapper v2.2.14 supporting librealsense v2.35.2 now published. Could you please have a try to see if the issue still existing for you? Thanks!

RealSenseSupport, 2020-07-10:
@furushchev  Did you get chance to try the latest version? Looking forward to your update. Thanks!

doronhi, 2020-07-27:
both realsense2_camera and librealsense2 are now available on the rosdistro DB. Usually they are a little behind the release of the official versions but they are there. Therefor I think this issue can now be closed.

closed 2020-07-27 by doronhi