realsenseai/realsense-ros #812: missing dependency with ddynamic_reconfigure

tanismar, 2019-06-11:
on the last updates, the dependency folder ddynamic_reconfigure has changed the name to realsense2_description. However, the CMakeLists.txt on those updates (tags 2.2.5 and later) still link to ddynamic_reconfigure (lines 18 and 76), which creates the following error:
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "ddynamic_reconfigure" with any of the following names:

    ddynamic_reconfigureConfig.cmake
    ddynamic_reconfigure-config.cmake

Maybe it passes the test based on some cached build, but this is clearly a broken dependency link.

doronhi, 2019-06-11:
Hi,
A couple of changes happened.
1. The folder ddynamic_reconfigure was removed in favor of the newly released debian package ddynamic_reconfigure.
2. The folder realsense2_camera was split to realsense2_camera and realsense2_description, for reducing the package basic size.

Therefor, the fix for the error is
`
apt-get update
apt-get install ros-kinetic-ddynamic-reconfigure
`

mblantonjr, 2019-06-11:
@doronhi 
running sudo apt-get install ros-kinetic-ddynamic-reconfigure
I get the following error:
Err:1 http://packages.ros.org/ros/ubuntu xenial/main amd64 ros-kinetic-ddynamic-reconfigure amd64 0.2.0-0xenial-20190325-154841-0800
  404  Not Found [IP: 64.50.233.100 80]
E: Failed to fetch http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-ddynamic-reconfigure/ros-kinetic-ddynamic-reconfigure_0.2.0-0xenial-20190325-154841-0800_amd64.deb  404  Not Found [IP: 64.50.233.100 80]

mblantonjr, 2019-06-11:
@doronhi 
Further info:
When running realsense-viewer, the app comes up, but indicates an error from the camera:
incomplete video frame detected. (see attached png)

Also, it appears that the firmware version is behind:
Current firmware version: 05.09.02.00
Minimal firmware version: 05.10.03.00

I am running:
[image]
[image]

Ubuntu 16.04LTS 64-bit
Intel® Core™ i7-6700TE CPU @ 2.40GHz × 8 
Intel® HD Graphics 530 (Skylake GT2) 
8GB memory

When running the following commands, I get...
command:  modinfo uvcvideo | grep "version:"

version:        1.1.2.realsense-1.3.5
srcversion:     3286F64D5B1431E64E8C37C

command:  sudo apt-get update
Ign:1 http://dl.google.com/linux/chrome/deb stable InRelease
Hit:2 http://dl.google.com/linux/chrome/deb stable Release                     
Hit:3 http://us.archive.ubuntu.com/ubuntu xenial InRelease                     
Hit:4 http://archive.canonical.com/ubuntu xenial InRelease                     
Get:5 http://security.ubuntu.com/ubuntu xenial-security InRelease [109 kB]     
Get:7 http://us.archive.ubuntu.com/ubuntu xenial-updates InRelease [109 kB]    
Hit:8 http://packages.osrfoundation.org/gazebo/ubuntu-stable xenial InRelease  
Get:9 http://packages.ros.org/ros/ubuntu xenial InRelease [4,678 B]            
Hit:10 http://linux.teamviewer.com/deb stable InRelease                        
Hit:11 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease
Hit:12 http://linux.teamviewer.com/deb preview InRelease                       
Hit:13 http://us.archive.ubuntu.com/ubuntu xenial-backports InRelease          
Err:9 http://packages.ros.org/ros/ubuntu xenial InRelease                      
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
Fetched 223 kB in 1s (186 kB/s)
Reading package lists... Done
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros/ubuntu xenial InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Failed to fetch http://packages.ros.org/ros/ubuntu/dists/xenial/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F42ED6FBAB17C654
W: Some index files failed to download. They have been ignored, or old ones used instead.

mblantonjr, 2019-06-12:
Additional update:
I upgraded the firmware on the D435 to 05.11.01.100 and it is working in the RealSense Viewer now, as is the T265.

Only remaining issue is getting the ROS package to build, which is still reporting the issue with the key not matching.

Any assistance is appreciated. Have a blessed evening.

doronhi, 2019-06-12:
A new GPG keys were deployed for packages.ros.org.
Follow the instructions [here](https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454)
In short, do the following:
`sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654`
`sudo apt-key del 421C365BD9FF1F717815A3895523BAEEB01FA116`
And now try 
`
sudo apt-get update
`
again.
After that install ddynamic_reconfigure package.

ManMan88, 2019-06-12:
And if I'm using ros melodic on Ubunto 18.04.2?
This package used to work with melodic, but now the ddynamic_reconfigure is missing and I can't find it for melodic distribution.

doronhi, 2019-06-12:
See #790

mblantonjr, 2019-06-12:
@doronhi 
Thank you for the updated key. Still getting an error on update:
W: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/aptrepo xenial Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/aptrepo/dists/xenial/main/binary-amd64/Packages  403  Forbidden
E: Some index files failed to download. They have been ignored, or old ones used instead.

ARLVOL, 2019-06-17:
I am getting that error too and I did your steps.
`CMake Error at /home/arl/catkin_ws/devel/share/ddynamic_reconfigure/cmake/ddynamic_reconfigureConfig.cmake:113 (message):
  Project 'ddynamic_reconfigure' specifies
  '/home/arl/catkin_ws/src/realsense/ddynamic_reconfigure/include' as an
  include dir, which is not found.  It does neither exist as an absolute
  directory nor in
  '/home/arl/catkin_ws/src/realsense/ddynamic_reconfigure//home/arl/catkin_ws/src/realsense/ddynamic_reconfigure/include'.
  Report the problem to the maintainer 'Noam Dori <theNODO55@gmail.com>' and
  request to fix the problem.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  realsense/realsense2_camera/CMakeLists.txt:8 (find_package)
`

doronhi, 2019-06-18:
The message you posted states "Report the problem to the maintainer 'Noam Dori <theNODO55@gmail.com>".
It seems that you have cloned ddynamic_reconfigure from a wrong source. If you indeed in a need to build that package yourself and not use apt-get install, try cloning the package from [here](https://github.com/pal-robotics/ddynamic_reconfigure/tree/kinetic-devel)

tanismar, 2019-06-19:
Hi @dorodnic , 
I followed your steps by installing ddynamic_reconfigure package, and checking out tag 2.2.6, and run into the same error as ARLVOL. If it is installed from a wrong source, means that the default source is wrong. Can this be fixed? thanks!

ARLVOL, 2019-06-19:
Yes,I cloned it by myself and not by apt-get and it is work.
It will be better to update it from the apt and not clone it every time.
tnx

PhilippeRerole, 2019-06-24:
I am also running ROS melodic on ubuntu 18.04.2, tried to build the ROS node source 2.2.6, ran into the same issue (ddynamic_reconfigure was missing from catkin_ws/src), cloned it from https://github.com/pal-robotics/ddynamic_reconfigure, and everything build smoothly then. Thanks for the help here.

doronhi, 2019-06-26:
> Hi @dorodnic ,
> I followed your steps by installing ddynamic_reconfigure package, and checking out tag 2.2.6, and run into the same error as ARLVOL. If it is installed from a wrong source, means that the default source is wrong. Can this be fixed? thanks!

What do you mean by "default source"? The realsense2_camera does not state a source for ddynamic_reconfigure.

ConnorChristie, 2019-09-08:
You can also try deleting the `build` folder in your catkin_ws then re-running `catkin_make`. This worked for me

zblcm, 2019-09-27:
I have the same issue and do not know how to solve ...

@doronhi 
Hi, if you see [realsense-ros/realsense2_camera/CMakeLists.txt](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/CMakeLists.txt) it shows "ddynamic_reconfigure" is required now in line 18. Should I simply remove that requirement, since you said reansense does not rely on it long time ago?

@PhilippeRerole 
Hi, did you do any extra operation after cloned and installed ddynamic_reconfigure from gihub?
I have cloned source, compiled then installed, but the same error pops again.

Thanks guys!

doronhi, 2019-10-06:
Sorry for misleading anyone. I didn't mean that "ddynamic_reconfigure" is not needed. Simply that it is not built by realsense2_camera and that the source code for that package is not included or pointed to by realsense2_camera.

The package "ddynamyic_reconfigure" is needed for building and running realsense2_camera.

RealSenseCustomerSupport, 2019-11-21:
----------------------------------------------

Still need any further clarification for this topic? If not, will close this.

mblantonjr, 2019-11-21:
No further clarification needed. Unit is operational.

Blessings,

Michael Blanton, Jr.

closed 2019-11-22 by RealSenseCustomerSupport

yuanzhedong, 2020-03-04:
I'm getting the same error on ubuntu 18.04, so the package is not available yet, right?

doronhi, 2020-03-08:
The package is available for ubuntu 18.04. type:
```
sudo apt-get install ros-melodic-ddynamic-reconfigure
```

sqlartist, 2021-09-06:
is it avaialable for windows?

dnovischi, 2022-12-27:
The lack of this package up-to-date source breaks all ros1 installs done from source. Hence, the issue at hand should be open. While one can compile using the source [here](https://github.com/pal-robotics/ddynamic_reconfigure). The package has not been updated or known to work since ros kinetic.

This renders Realsense's useless in real world robotic scenarios. For example, using the Jetson Nano for inference with Realsense. That is the Nano official image has Ubuntu 18.04 and running native ros1 requires installation from source. Other hacks are possible (like docker, upgrading manually) none of which will work with realsense (at least not with 435i or 455).

MartyG-RealSense, 2022-12-27:
Hi @dnovischi  ddynamic_reconfigure can be installed using **apt-get install** with specific versions for Kinetic, Melodic and Noetic.

Kinetic
sudo apt install ros-kinetic-ddynamic-reconfigure

Melodic
sudo apt install ros-melodic-ddynamic-reconfigure

Noetic
sudo apt install ros-noetic-ddynamic-reconfigure

dnovischi, 2022-12-30:
Hi @MartyG-RealSense, I don't understand how having having some packages solves the issue.

The setup is the following:

1. Jetson Nano with Official Ubuntu 18.04 installed:

```bash
$ cat /etc/os-release 
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
```

2. Ros Noetic (full desktop) compiled and installed from source using official instructions [here](http://wiki.ros.org/noetic/Installation/Source). The installation was checked and is working flawlessly (with default python3.6) on Ubuntu 18.04 official Jetson Nano image:

```bash
$ roscore
... logging to /home/jetson/.ros/log/56d5b7c4-8844-11ed-873c-ae8a2019ff15/roslaunch-jetson-antrobot3.local-26833.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-antrobot3.local:38921/
ros_comm version 1.15.15

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES

auto-starting new master
process[master]: started with pid [26843]
ROS_MASTER_URI=http://jetson-antrobot3.local:11311/

setting /run_id to 56d5b7c4-8844-11ed-873c-ae8a2019ff15
process[rosout-1]: started with pid [26856]
started core service [/rosout]
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
```

4. Librealsense 2.50.0 compiled from source using official instructions [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md). And udev rules checked to see if they were installed properly.

5. Install compatible Realsense 435i firmware version 5.13.0.50 as specified  officially [here](https://dev.intelrealsense.com/docs/firmware-releases). Firmware checked with (x86 PC) realsense-viewer and ros noetic instalation via rviz, rqt and command line and is working properly.

6.  Cloned the realsense-ros repo and checkout ros1-legacy with compatible tag version 2.3.2. Installed the only version of ddynamic_reconfigure installable possible on ubuntu18.04 : `sudo apt install ros-melodic-ddynamic-reconfigure` (for with I'm sorry, but I fail to see the relevance of since we're talking about ros-noetic installation through source).

7. Tried to compile the realsense-ros using the official steps [here](https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy) which of course give:

```bash
$ catkin_make clean
Base path: /home/jetson/catkin_ws
Source space: /home/jetson/catkin_ws/src
Build space: /home/jetson/catkin_ws/build
Devel space: /home/jetson/catkin_ws/devel
[... 75 lines cut ...]

Now as per my official post I have already tried to use the source available [here](https://github.com/pal-robotics/ddynamic_reconfigure/tree/kinetic-devel) suggested in the official compilation steps. The compilation succeeds but nothing works.

I have also tried to install the ros-noetic ddynamic-reconfigure package: `sudo apt install ros-noetic-ddynamic-reconfigure` which, of course,  is not installable on ubuntu 18.04:

```bash
 sudo apt install ros-noetic-ddynamic-reconfigure
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-noetic-ddynamic-reconfigure : Depends: libgcc-s1 (>= 3.0) but it is not installable
                                   Depends: libstdc++6 (>= 9) but 8.4.0-1ubuntu1~18.04 is to be installed
                                   Depends: ros-noetic-dynamic-reconfigure but it is not going to be installed
                                   Depends: ros-noetic-roscpp but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

**So, @MartyG-RealSense what am I missing? What is the solution for installations from source?**

MartyG-RealSense, 2022-12-30:
In the RealSense ROS1 wrapper, ddynamic_reconfigure is a required component.  It allows changes to settings to be made dynamically during runtime after launch instead of in a launch file or roslaunch instruction before launch.  It is not required in the ROS2 wrapper.

ddynamic_reconfigure can be installed by cloning its source as the wrapper installation instructions suggest.  Whilst it is certainly valid to install it this way, I personally always recommend installing it from packages instead by using the apt-install packages for particular ROS1 branches that I listed above.  The reason for this recommendation is simply because it is an easy installation method.

Checking carefully through your detailed description of your complete installation procedure on Jetson, I cannot see anything that is obviously incorrect and you have installed the various components in the appropriate order (librealsense, ROS, ddynamic_reconfigure, RealSense ROS1 wrapper).  Your librealsense / Python installation is also working correctly.

There are occasionally problems with the ROS1 wrapper that are not correctable no matter what is tried, but are suddenly resolved if the entire **catkin_ws/src** catkin workspace folder is deleted and the wrapper is built from source again.

dnovischi, 2022-12-30:
>  In the RealSense ROS1 wrapper, ddynamic_reconfigure is a required component. It allows changes to settings to be made dynamically during runtime after launch instead of in a launch file or roslaunch instruction before launch. It is not required in the ROS2 wrapper.

So the best bet when using Jetson Nano is to do a distribution upgrade to ubuntu 20.04 and switch to ROS2?
Does the same apply to Jetson Xavier NX?

> There are occasionally problems with the ROS1 wrapper that are not correctable no matter what is tried, but are suddenly resolved if the entire catkin_ws/src catkin workspace folder is deleted and the wrapper is built from source again by following the instructions from the beginning after the catkin_ws/src folder is recreated. Choose to either install ddynamic_reconfigure from source or packages but do not use both methods.

I'll give it a try. Also can you post your way of re-configuring udev rules?

On the other hand, what's your recommended way of using Realsense (435i and/or 455) with Jetson Nano based on a python3.x version? (I really don't care about the OS, as long as I can run ROS on python3. Our code requires tensorrt compiled models to run inside of ros nodes).

Thanks and Happy New Year!

PS: I'll post findings for this in more appropriate issue, as continuing the discussion here is kind of out of scope. (Probably in #2580)

MartyG-RealSense, 2022-12-30:
You are very welcome!

A RealSense user provides a complete guide at https://github.com/IntelRealSense/realsense-ros/issues/2197#issuecomment-1193176709 for using Python 3 with ROS, Jetson Nano and RealSense D455.