realsenseai/realsense-ros #813: Any realsense-ros branch for ROS-kinetic?

HappySamuel, 2019-06-12:
Hi

I see a lot of branch, but i couldn't find a branch specifically for ROS-kinetic. Is there any release branch for kinetic?

Best,
Samuel

doronhi, 2019-06-12:
The main branch is for kinetic. There is no specific branch for anything else yet.

HappySamuel, 2019-06-12:
Hi @doronhi 

The branch "development" is for ROS-kinetic, is this what you mean?

Best,
Samuel

doronhi, 2019-06-12:
That's what I mean.
You can either pull from there or from the latest release.

HappySamuel, 2019-06-12:
Hi @doronhi 

Thank your for the information.

Best,
Samuel

HappySamuel, 2019-06-12:
Hi @doronhi 

I just tried the development branch and i got the error as shown as below when running it.

[image]

Is there a way to solve it?

Best,
Samuel

doronhi, 2019-06-12:
Its a recurring issue in librealsense (search in [librealsense issues](https://github.com/IntelRealSense/librealsense/issues))
The recommended action is to disconnect and reconnect the camera or to run the realsense2_camera node with flag `--initial_reset:=true`

HappySamuel, 2019-06-13:
Hi @doronhi 

I tried both your suggestions, but i still got the same error as above. Does librealsense currently has a solution for this?

Best,
Samuel

doronhi, 2019-06-13:
What happens when you use realsense-viewer?

HappySamuel, 2019-06-19:
Hi @doronhi 

This is the screenshot of using realsense-viewer.
[image]

Best,
Samuel

HappySamuel, 2019-06-19:
Hi @doronhi 

A quick update, after i upgrade the d435 firmware, the issue above gone. Now i can view it in both realsense viewer and RVIZ. I am following the below upgrade instruction from this website:
https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Linux-RealSense-D400-DFU-Guide.pdf

Best,
Samuel

closed 2019-06-19 by HappySamuel