realsenseai/realsense-ros #816: Compiling issue

mpkuse, 2019-06-17:
Hi, 
I have a realsense d435i. I followed https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md and could install the drivers and dev packages. 

I can confirm that the `realsense-viewer` works as expected, I could see the Depth data, RGB as well as motion data. Next I also tried to compile `hello-realsense` example (https://github.com/IntelRealSense/librealsense/tree/development/examples/hello-realsense) like `g++ -std=c++11 rs-hello-realsense.cpp  -lrealsense2` that also works as expected. 

However, I have issues compiling the realsense-ros. 
```
dji@MANIFOLD-2-C:~/catkin_ws/src/dev_drivers/realsense-ros$ git status 
HEAD detached at 2.2.6
nothing to commit, working directory clean
```

catkin_make the workspace gives: 
```
In file included from /home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:7:0,
                 from /home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1:
/home/dji/catkin_ws/src/dev_drivers/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:213:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic push // CLion suppressor
 ^
/home/dji/catkin_ws/src/dev_drivers/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:214:0: warning: ignoring #pragma ide diagnostic [-Wunknown-pragmas]
           #pragma ide diagnostic ignored "OCUnusedGlobalDeclarationInspection"
 ^
/home/dji/catkin_ws/src/dev_drivers/ddynamic_reconfigure/include/ddynamic_reconfigure/ddynamic_reconfigure.h:220:0: warning: ignoring #pragma clang diagnostic [-Wunknown-pragmas]
           #pragma clang diagnostic pop
 ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)’:
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:281:22: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘registerVariable’
             ddynrec->registerVariable<bool>(
                      ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:281:39: error: expected primary-expression before ‘bool’
             ddynrec->registerVariable<bool>(
                                       ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:283:89: error: expected primary-expression before ‘,’ token
               [option, sensor](bool new_value) { sensor.set_option(option, new_value); },
                                                                                         ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:309:24: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘registerVariable’
               ddynrec->registerVariable<int>(
                        ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:309:41: error: expected primary-expression before ‘int’
               ddynrec->registerVariable<int>(
                                         ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:311:92: error: expected primary-expression before ‘,’ token
                   [option, sensor](int new_value) { sensor.set_option(option, new_value); },
                                                                                            ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:329:28: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘registerVariable’
                   ddynrec->registerVariable<double>(
                            ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:329:45: error: expected primary-expression before ‘double’
                   ddynrec->registerVariable<double>(
                                             ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:331:99: error: expected primary-expression before ‘,’ token
                       [option, sensor](double new_value) { sensor.set_option(option, new_value); },
                                                                                                   ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:357:22: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘registerEnumVariable’
             ddynrec->registerEnumVariable<int>(
                      ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:357:43: error: expected primary-expression before ‘int’
             ddynrec->registerEnumVariable<int>(
                                           ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:359:90: error: expected primary-expression before ‘,’ token
                 [option, sensor](int new_value) { sensor.set_option(option, new_value); },
                                                                                          ^
/home/dji/catkin_ws/src/dev_drivers/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:363:14: error: ‘class ddynamic_reconfigure::DDynamicReconfigure’ has no member named ‘publishServicesTopics’
     ddynrec->publishServicesTopics();
              ^
dev_drivers/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:86: recipe for target 'dev_drivers/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o' failed
make[2]: *** [dev_drivers/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
CMakeFiles/Makefile2:5635: recipe for target 'dev_drivers/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [dev_drivers/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

```

my ddynamic_configure package: 
```
dji@MANIFOLD-2-C:~/catkin_ws/src/dev_drivers/ddynamic_reconfigure$ git status 
On branch dubnium-devel
Your branch is up-to-date with 'origin/dubnium-devel'.
nothing to commit, working directory clean
```

can you help it set this up please.

doronhi, 2019-06-17:
Branch "dubnium-devel" is very old. It seems that the source you have for ddynamic_reconfigure is not up to date. See #805
So, first thing - remove your copy of ddynamic_reconfigure.
If you are using kinetic, the simple thing would be to `apt-get update` and `apt-get ros-kinetic-ddynamic-reconfigure`.
If not, install from source [here](https://github.com/pal-robotics/ddynamic_reconfigure)

mpkuse, 2019-06-17:
Perfect!!

closed 2019-06-17 by mpkuse

ChaiZQ, 2019-06-18:
wonderful!