realsenseai/realsense-ros #817: Customizing roslaunch rs_camera.launch

mpkuse, 2019-06-17:
Hi, 
I have a working realsense ros package. I could verify the camera stream using rqt_image_view. 
I copied the `rs_camera.launch` in my ros package. I am wanting to use the stereo camera aka infra1 and infra2. 

On the image (with topic /camera/infra2/image_rect_raw) I see the dots which I assume are from the infrared projector. How can I turn this off ? 

[image]

doronhi, 2019-06-18:
You can add the following lines to your launch file, before launching realsense2_camera node:
```
<rosparam>
  /camera/stereo_module/emitter_enabled: false
</rosparam>
```

mpkuse, 2019-06-18:
Perfect., works!

closed 2019-06-18 by mpkuse

quxichao, 2019-06-20:
Hi,I want to do the same thing with you.I want to turn off the IR projector by the rs_camera.launch.But I am a newcomer to the ROS platform.I don't know where to add this code. Can you help me?

<rosparam>
  /camera/stereo_module/emitter_enabled: false
</rosparam>

Thank you!

vhanded, 2019-06-23:
Sorry for very new to ROS. I set the rosparam into nodelet.launch.xml, and it works.

But I would like to toggle the emitter easily, is it possible to set in command, like this?

`roslaunch realsense2_camera rs_camera.launch /camera_stereo_module/emitter_enabled:=false`

?

doronhi, 2019-06-25:
You could create your own launch file, based on rs_camera.launch and add the following lines:

```
  <arg name="emitter_enable"   default="false"/>

<!-- rosparam set /camera/stereo_module/emitter_enabled false -->
<rosparam if="$(arg emitter_enable)">
  /camera/stereo_module/emitter_enabled: true
</rosparam>
```

Then you could run 
`roslaunch realsense2_camera rs_camera.launch emitter_enable:=true`
Does that work for you?

AndreV84, 2021-04-03:
> You could create your own launch file, based on rs_camera.launch and add the following lines:
> 
> ```
>   <arg name="emitter_enable"   default="false"/>
> 
> 
> <rosparam if="$(arg emitter_enable)">
>   /camera/stereo_module/emitter_enabled: true
> </rosparam>
> ```
> 
> Then you could run
> `roslaunch realsense2_camera rs_camera.launch emitter_enable:=true`
> Does that work for you?

it doesn't work anymore?  does it? how to just disable the emitter in a simpe way?

doronhi, 2021-04-05:
emitter_enabled is an ENUM now, so we need to specify 0 instead of false. (you can see available values using rqt_reconfigure).
use the following lines in the launch file to turn the emitter off:
```
<rosparam>
  /camera/stereo_module/emitter_enabled: 0
</rosparam>
```

AndreV84, 2021-04-05:
@dorodnic 
with the GUI opened by the commadn below I do not see th eemitter option listed at all
```
rosrun  rqt_reconfigure  rqt_reconfigure
```
is it the correct way to open the GUI to see the emitter option?
[image]

AndreV84, 2021-04-05:
found it; thanks

w407022008, 2021-09-21:
> found it; thanks

Hello Andre, how did you find it in GUI? I hope to know more other params.Thx a lot.

AndreV84, 2021-09-21:
@w407022008 yes it is presented in some of entries, as far as I recollect [ I do  not have it at hand right now to check with though]

w407022008, 2021-09-21:
Oh yes, it is presented when i recollect it again. It looks different from before. Probably one of params has been set in lauch file, and those will be then shown. Thx again.

gsanya, 2021-10-05:
If someone is still interested:
For me this worked to turn off the emitter:
`<rosparam> /camera/stereo_module/emitter_enabled: 2</rosparam>`
I put it in my own launch file after starting the camera lik this:
'    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
		<arg name="enable_gyro" value="true" />
		<arg name="enable_accel" value="true" />
		<arg name="unite_imu_method" value="linear_interpolation" />
		<arg name="color_width" value="640" />
		<arg name="color_height" value="480" />
		<arg name="color_fps" value="30" />
		<arg name="enable_infra1" value="true" />
		<arg name="enable_infra2" value="true" />
		<arg name="infra_width" value="640" />
		<arg name="infra_height" value="480" />
		<arg name="infra_fps" value="30" />
    </include>
	<rosparam> /camera/stereo_module/emitter_enabled: 2</rosparam>'
If I understand correctly, now the emitter_enabled param is an enum, and 2 means off. This param doesn't exists before setting it like this, so you will only find It in the GUI, after creating it in some lounch file.