realsenseai/realsense-ros #818: Camera info distortion model always set to "blumb_bob"

AlexReimann, 2019-06-17:
The distortion model in the camera is not correctly set but rather is fixed to `blumb_bob`:

https://github.com/IntelRealSense/realsense-ros/blob/de1649a04ff7f96b549773fc709b3f8b2a81d397/realsense2_camera/src/base_realsense_node.cpp#L1639

Does this have some compatibility reasons and can be expected to fix in the future?

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Hi @AlexReimann

No planned changes to this distortion model.

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

I'll be closing this thread out if there is nothing further necessary.

closed 2020-03-17 by RealSenseCustomerSupport

AlexReimann, 2020-03-18:
The camera reports actually different model than `plumb_bob` and this is actually available in this context.
See also:

https://github.com/IntelRealSense/librealsense/blob/1d7ef6ac04394d6183178f6e2dd9304d34bb44ff/include/librealsense2/h/rs_types.h#L44-L55

doronhi, 2020-03-19:
@AlexReimann , thanks for the tip. I'll look into it.