realsenseai/realsense-ros #820: Hardware failures after prolonged use

MrCerealKiller, 2019-06-19:
# Persistent Hardware Failures even on USB 3.2

When using realsense-ros, after a sometimes very short amount of time, I get hardware warnings and subsequent failures. The application I'm making would require the device to be used over extended periods of time, but I usually can't go one minute before encountering this.

| System Info | |
| -- | -- |
| OS | Ubuntu 18.04 |
| ROS Distro | Melodic |
| ibrealsense Version | 2.22.0 |
| realsense-ros Version | development |
| firmware version | 5.11.6.250 |

| Setup Info | |
| -- | -- |
| Product | D435i |
| USB Interface | 3.2 |
| Resolution (All) | 640x480 |
| FPS (All) | Tried 20/30/60 |

So I realize that this may be a librealsense issue, but I only seem to encounter this issue while using it with the ros wrapper (i.e. it seems to work fine using `realsense-viewer`). I also understand that melodic isn't officially supported yet, but I'm hoping for any guidance so that I can solve the problem myself.

Essentially, after prolonged use, I always get the following warning:
```
[ WARN] [____]: Hardware Notification:USB SCP overflow,____,Error,Hardware Error
```
To attempt to remedy this I've tried lowering the framerate and even eliminating one of the infra streams. I still get the warning eventually, even though I don't think this should be an issue...

Shortly after, I get an actual error:
```
ERROR [____] (backend-v4l2.h:207) xioctl(VIDIOC_QBUF) guard failed
```

Below is a full log of what happens when I try to use the camera. A couple of things to note though:
* Here I'm using librealsense 2.23.0 because I thought maybe the latest deb release for bionic would have the fix. I encounter the issue on both 2.23.0 and 2.22.0
* I don't have this hardware patch that it mentions... I found what seem to be the patches in the librealsense-dkms, but they are Xenial specific and I'm not sure if they still apply (this was another reason I updated to 2.23.0)
```

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 60
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 60
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
[... 118 lines cut ...]
 19/06 15:46:59,526 ERROR [140250163218176] (backend-v4l2.h:207) xioctl(VIDIOC_QBUF) guard failed
 19/06 15:46:59,526 ERROR [140250163218176] (backend-v4l2.h:207) xioctl(VIDIOC_QBUF) guard failed
 19/06 15:46:59,526 ERROR [140250163218176] (backend-v4l2.h:207) xioctl(VIDIOC_QBUF) guard failed
[ WARN] [1560973619.721908718]: Hardware Notification:USB SCP overflow,1.56097e+12,Error,Hardware Error
 19/06 15:47:04,922 WARNING [140250584393472] (types.cpp:48) get_xu(ctrl=1) failed! Last Error: Input/output error
 19/06 15:47:04,923 ERROR [140250584393472] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: get_xu(ctrl=1) failed! Last Error: Input/output error
[ WARN] [1560973625.723996451]: Hardware Notification:USB SCP overflow,1.56097e+12,Error,Hardware Error
[ WARN] [1560973629.725370882]: Hardware Notification:USB SCP overflow,1.56097e+12,Error,Hardware Error
 19/06 15:47:10,952 WARNING [140250584393472] (types.cpp:48) set_xu(ctrl=1) failed! Last Error: Input/output error
 19/06 15:47:10,952 ERROR [140250584393472] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Input/output error
[ WARN] [1560973636.727479508]: Hardware Notification:USB SCP overflow,1.56097e+12,Error,Hardware Error
 19/06 15:47:16,981 WARNING [140250584393472] (types.cpp:48) set_xu(ctrl=1) failed! Last Error: Input/output error
 19/06 15:47:16,982 ERROR [140250584393472] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Input/output error
[ WARN] [1560973638.728770005]: Hardware Notification:USB SCP overflow,1.56097e+12,Error,Hardware Error
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
^C^C 19/06 15:47:23,014 WARNING [140250584393472] (types.cpp:48) set_xu(ctrl=1) failed! Last Error: Input/output error
 19/06 15:47:23,014 ERROR [140250584393472] (global_timestamp_reader.cpp:165) Error during time_diff_keeper polling: set_xu(ctrl=1) failed! Last Error: Input/output error
terminate called without an active exception
shutting down processing monitor...
... shutting down processing monitor complete
done
```

**Any guidance would be very much appreciated!**

pyni, 2019-07-08:
have you solved problem?

MrCerealKiller, 2019-07-08:
@pyni No, I've been focusing on other parts of the project because it was hard to deal with

jpapon, 2019-09-21:
I have the same problem, with Librealsense 2.21-2.28.
I have it with both this implementation of a ros realsense nodelet, as well as my own, so I think it's a problem with librealsense itself.
The error "terminate called without an active exception" is common when threads aren't properly joined/detached, so I'm going to try fiddling with OMP and the number of threads the nodelet manager has to see if I can get it to stop happening.

For reference, to reproduce this bug, start a roscore in a separate terminal, then launch the realsense-ros nodelet in another terminal. Then kill/launch the realsense node a few times, and you will eventually get a situation where the nodelet just crashes with the message "terminate called without an active exception".

RealSenseCustomerSupport, 2020-01-16:
----------------------------------------------

Hi @MrCerealKiller,

Please provide us with the following details about your system configuration :

how many cameras are in active use in the system

build number of ROS Wrapper

HW configuration details (platforms names, how the camera(s) is/are connected, etc)

Along with that information, please describe the exact scenario that leads to the problem.

Thank you!

RealSenseCustomerSupport, 2020-01-22:
----------------------------------------------

@MrCerealKiller,

Do you still need help with this question?

closed 2020-01-29 by RealSenseCustomerSupport

JadTawil-theonly, 2020-04-16:
@jpapon I am having the same problem! Any thoughts?