realsenseai/realsense-ros #821: Intel RealSense SDK 2.0 is missing

lilala-lsc, 2019-06-20:
I install librealsense2 by https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md 
When I install realsense-ros as https://github.com/IntelRealSense/realsense-ros, I got a problem.
`catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
`
[image]
But on another computer, it have no problem.

phil-ludewig, 2019-06-20:
Got the same issue here.
Using a Jetson Nano and installed the librealsense SDK via the [JetsonHacks Script](https://www.jetsonhacks.com/2019/05/16/jetson-nano-realsense-depth-camera/) + patched kernel.
Using ROS melodic, would love to see a solution.

joekeo, 2019-06-24:
Same here, Ubuntu 18.04, ROS Melodic

phil-ludewig, 2019-06-24:
Found a solution to this! You'll have to clone librealsense (not librealsense-ros) into your catkin_ws/src before cloning anything else. Then build librealsense according to instructions.

Next, also clone realsense- ros & ddynamic_reconfigure into your /src. Run rosdep update in /src folder. Go back to catkin_ws and install dependencies.
DON'T CATKIN_MAKE YET.

Now the dependencies for librealsense have been added, but before catkin_make you have to remove the librealsense (not a catkin_package) from your src folder. Then run catkin_make according to librealsense-ros instructions. No need to put librealsense back into src.

Worked perfectly for me.

joekeo, 2019-06-24:
Many thanks for the tip!
I will definitely will try this.

The problem is that then I will have to do the same with the other 4 robots... and when a proper fix rolls out, do it again.

doronhi, 2019-06-26:
I'm glad it worked out for you but I'm concerned it may be a problematic solution.
You originally tried to install librealsense using debian packages and then reverted to cloning and compiling the code.
This suggest that something wrong happened with the original installation.
Even compiling librealsense from code, there is no requirement for it to happen inside catkin_ws.
You should check now that you don't have another copy of librealsense somewhere, maybe of a different version.

lilala-lsc, 2019-06-26:
I have got the solution. I check the **realsense-ros/.travis.yml** and find it need **librealsense2-dkms**. 
`sudo apt-get install librealsense2-dev`

closed 2019-06-26 by lilala-lsc

phil-ludewig, 2019-06-26:
@doronhi: Sorry, I didn't mention that I wiped my system after installing librealsense with the JetsonHacks script. From JetsonHacks I only used the Kernel patch script.
Haven't tried a librealsense cmake with the package outside of catkin_ws/src, but probably works just as well.

@lilala-lsc: Awesome, will try it whenever I have to reinstall :D

caffreyu, 2019-11-02:
> I have got the solution. I check the **realsense-ros/.travis.yml** and find it need **librealsense2-dev**.
> `sudo apt-get install librealsense2-dev`

I still cannot figure this out. Could you please give me more details about how did you install the realsense-ros? Did you follow the procedure provided by JetsonHacks?

Thanks in advance!

vishnuu95, 2020-08-08:
One issue could be that realsense-ros is looking for realsense2 2.37.0 but the version of librealsense installed is only 2.36.0, even though I built the latest version of librealsense. This can be fixed by just editing the CMakeLists.txt found at <local_path>/realsense_ros/realsense2_camera/CMakeLists.txt

Fred159, 2020-08-10:
> One issue could be that realsense-ros is looking for realsense2 2.37.0 but the version of librealsense installed is only 2.36.0, even though I built the latest version of librealsense. This can be [...]

awesome~ works like magic

ScottMcCormack, 2020-08-10:
Looks like version [`2.2.16`](https://github.com/IntelRealSense/realsense-ros/tree/2.2.16) updated it to use realsense2 v2.37.0

Reverting back to version [`2.2.15`](https://github.com/IntelRealSense/realsense-ros/tree/2.2.15) worked for me

`git clone https://github.com/IntelRealSense/realsense-ros.git --branch 2.2.15`

StevenLiuMaster, 2024-05-17:
> I have got the solution. I check the **realsense-ros/.travis.yml** and find it need **librealsense2-dev**. `sudo apt-get install librealsense2-dev`

brilliant job, well done