realsenseai/realsense-ros #822: Set intrinsic matrix manually

sean85914, 2019-06-21:
I used an apriltag to do **hand-eye calibration** between my robot arm and the camera (which is a **SR300**). I noticed that the position obtained from apriltag detector tended to "**move down**" from the real tag center which made my calibration result inaccurate. I thought the reason is came from **bad intrinsic parameters** but it seems that there is not any service for me to set the parameters. I wonder if there is any method to do so?

doronhi, 2019-06-26:
The intrinsics include focal length and principal point. I'm not sure that that's what affecting your calibration..
You can check your intrinsics calibration using `rs-depth-quality-tool`.

However, to answer your question, there is no method to fix intrinsic calibration on SR300. Not in the librealsense2 API and not in the realsense2_camera wrapper.

On D400 series there is a dynamic calibration tool. You can check it out [here](https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API) 

You could write your own wrapper (Starting from realsense2_camera if it suits you). Using librealsense2 API you could define a "software device" and set it to whatever intrinsics parameters you want. That, however, is something I didn't try and altogether out of the scope of this repository.

sean85914, 2019-06-26:
Thanks you reply. Does `rs-depth-quality-tool` can only calibrate depth camera? Or it can also calibrate color camera?

doronhi, 2019-06-27:
As far as I know, the tool cannot calibrate at all. It simply shows the accuracy of the depth and by extension, the correctness of the intrinsic parameters.

sean85914, 2019-07-01:
Sure, the output of `rs-depth-quality` intrinsic matrix seems the same with `/camera/depth/camera_info`. I will do calibration with calibration board and compare the result with `/camera/color/camera_info` ASAP

npd26, 2019-07-18:
Hi @sean85914 have you found out how to set matrix manually in ROS topic?

npd26, 2019-07-23:
@sean85914 hi, is there any update?

sean85914, 2019-07-23:
Hi @npd26, sorry for the late reply, since the prior hand-eye calibration result were acceptable. 
Since I changed my setup today, I have do the procedure once again.
For manually setting the camera info of the realsense camera, I found that you can use the method as following:

1. Manually calibrate the color camera by yourself and record the data, this can be done using `cameracalibrator.py` from `camera_calibration` [[tutorial]](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration)
2. Remap the topic `/camera/color/camera_info` to another name, say `/foo` (you have to revise the launch filepackage://realsense2_ros/launch/includes/nodelet.launch.xml, just add a remap tag)
3. Write a node which subscribe `/foo` and publish `/camera/color/camera_info. Put the matrices you recorded into the topic you will publish, for others information, just keep as default.

While I noticed that after I changed the size of the tag, the result became much better than manually setting the intrinsic matrices. :sweat_smile:

waiyc, 2019-08-02:
@sean85914  For the SR300 camera, may I know how do you calibrate the camera using opencv calibration package? 
Do you need to calibrate the depth image topic again by using monocalibration after calibrate the IR cameras as stereo?

-chan

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

This thread has been open for a while without any response.  I think it can be closed at this time.

closed 2020-02-05 by RealSenseCustomerSupport