realsenseai/realsense-ros #824: under what circumstances should I turn on the flag of enable_sync?

xhuan28, 2019-06-24:
There is an argument called ["enable_sync"](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/includes/nodelet.launch.xml#L41) in [nodelet.launch.xml](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/includes/nodelet.launch.xml) which is set to false.  
I am wondering under what circumstances it should be set to true. How should I use this argument?
Thanks.

samuelclay, 2019-06-25:
Use the sync argument when you have multiple cameras hooked up together so that the infrared dots are only displayed one at a time. 

See https://www.intel.com/content/www/us/en/support/articles/000028140/emerging-technologies/intel-realsense-technology.html for more information.

doronhi, 2019-06-26:
`enable_sync` causes the images to run though a "syncer" before being published. 
It helps to ensure that frames from different streams, taken at about the same time, will arrive as close together as possible.
For example, if you have a depth stream of very low resolution it's processing time will be short. If you have in addition a color stream with very high resolution it's processing time will be much longer. If they are both taken together, the depth one will be published a while before the color one. 
The syncer basically holds the depth image until the color image is ready too and publishes them together.
It can be useful if you want to implement pointcloud or image alignment by yourself or something else that need several streams synchronized.

xhuan28, 2019-06-27:
@doronhi 
Frames in frameset from pipeline should be synchronized and if I set the same ros timestamp to each frame in a frameset before publishing, they should be treated as synchronized frames by other ros node as well. Sync mode doesn't introduce much more resource consumption.  Is my understanding correct? 
If that is the case, it is better to enable sync_mode always. Is there any side effect if I always enable sync mode, whenever getting a frameset or a single frame?

doronhi, 2019-06-27:
The downside is that you might drop frames in the synchronization. If the images are not being used together, like if you want to display each stream separately, you might loose some frames and gain nothing but enabling sync.

xhuan28, 2019-07-01:
@doronhi 
Got it. Thanks for your answers.

closed 2019-07-01 by xhuan28