realsenseai/realsense-ros #825: T265 random crashes

joekeo, 2019-06-24:
Hello,

I have been having a lot of issues lately with the T265 camera.
I am running it alongside a D435i.
this is how a normal startup looks like:
```
[23572] [E] Device-6190: ERROR: Bulk transfer message 0x25 (DEV_SET_LOW_POWER_MODE) request to device got LIBUSB_ERROR_IO. Bytes requested 8, bytes transferred 0
13:55:05.561 [23572] [E] Device-6190: ERROR: Bulk transfer message 0x1 (DEV_GET_DEVICE_INFO) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
13:55:05.561 [23572] [E] Device-6190: USB Error (0x6)
13:55:05.561 [23572] [E] Device-6190: Error: Failed to get device info (0x6)
13:55:05.563 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.565 [23572] [E] Device-6190: ERROR: Bulk transfer message 0x1 (DEV_GET_DEVICE_INFO) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
13:55:05.565 [23572] [E] Device-6190: USB Error (0x6)
13:55:05.565 [23572] [E] Device-6190: Error: Get Device Info error (0x6)
13:55:05.565 [23572] [E] Device-6190: Entered state [ERROR_STATE]
[INFO] [1561380905.567770]: Serving HTTP on 0.0.0.0 port 8000...
13:55:05.573 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
[INFO] [1561380905.576134]: Started controllers: joint_state_controller, mobile_base_controller
13:55:05.583 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.594 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.604 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.614 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.625 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.635 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.645 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.656 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
[INFO] [1561380905.662227]: Rosbridge WebSocket server started on port 9090
13:55:05.666 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.676 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.686 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.697 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.707 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.717 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.727 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.737 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.747 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.758 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.768 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.778 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.788 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.798 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.809 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.819 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.829 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.839 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.849 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
13:55:05.860 [23573] [E] Device-5B70: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)
 24/06 13:55:06,076 WARNING [140276732155648] (types.cpp:48) hwmon command 0x4f failed. Error type: No data to return (-21).
[ INFO] [1561380906.127202522]: Resetting device...
13:55:06.128 [23550] [E] Device-6190: Error: Got Temperature for 1391726288 sensors, exceeding max sensors (3)
 24/06 13:55:06,128 ERROR [140321317496576] (types.h:243) Failed to query T2xx tracking temperature option
terminate called after throwing an instance of 'rs2::error'
  what():  Failed to query T2xx tracking temperature option
[rs_t265/realsense2_camera_manager-12] process has died [pid 23438, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/rossrobo/.ros/log/e649cc2c-9673-11e9-abeb-94c69118d9e8/rs_t265-realsense2_camera_manager-12.log].
log file: /home/rossrobo/.ros/log/e649cc2c-9673-11e9-abeb-94c69118d9e8/rs_t265-realsense2_camera_manager-12*.log

```
or sometimes does the classic:
```
[ERROR] [1561381358.582125388]: The requested device with serial number 908412110861 is NOT found. Will Try again.
```

This is a normal startup. Sometimes it will fully initialise. After initialisation, the sensor performs well for a bit, then crashes ( taking the full navigaton stack with it).

Any idea how to fix these issues?

doronhi, 2019-06-25:
Sorry, I can't reproduce that. 
There are many error messages during initialization due to the nature of the T265 device, disconnecting and reconnecting again as a different device and due to it's locking the context for a while. This does make debugging harder.
So the messages "Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)" are normal. 
On the other hand, the messages about the temperature and device info are not.

- I see in #821 that you had a weird installation. The problem there raise the suspicion that you have more than one installation of librealsense2 on your device, so it could also be related. The beginning part of the realsense2_camera log should include the versions of librealsense2, realsense2_camera as well as the firmware versions of the devices. Could you include them please?

- You mentioned that you " have been having a lot of issues **lately**..." Was it working fine before? Anything changed? An upgrade or something?

- Is running realsense2_camera with each device separately works fine?

- What OS do you run on? What kind of machine?

joekeo, 2019-07-02:
The issues in (#821) are happening in another machine (robot) which I recently updated to the latest 2.2.6 tag and lib realsense.

The issues here are happening using the 2.2.4 tag of this repo and runnin in Ubuntu 18.94 ros melodic. 
I removed the D435i from the stack and now I am only running the T265. The issues here persist even with just one of the cameras. 

I used to have one camera on one platform that rarely had any issued, and the issues were only in startup. Then I bought this camera and after installing it in this robot, I have been having random crashes. But they were somewhat rare, recently (after updating to 2.2.4 tag and with the latest librealsense) there is always a crash, sometimes 3 minutes into the operation, sometimes 10 or 30. But it always crashes. When it crashes it takes down the full navigation/localisation stack.

Below is the most common error that I am having:
```
15:53:58.750 [1615] [E] Device-6860: ERROR: Bulk transfer message 0x25 (DEV_SET_LOW_POWER_MODE) request to device got LIBUSB_ERROR_IO. Bytes requested 8, bytes transferred 0
15:53:58.754 [1615] [E] Device-6860: ERROR: Bulk transfer message 0x1 (DEV_GET_DEVICE_INFO) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
15:53:58.754 [1615] [E] Device-6860: USB Error (0x6)
15:53:58.754 [1615] [E] Device-6860: Error: Failed to get device info (0x6)
```

Any idea what it might be the culprit of this issue?

joekeo, 2019-07-03:
The device keeps crashing after 5+ minutes every timne (sometimes more).
I have downgraded, purged and updated the librealsense some times now. 

This are some of the other error messages I keep getting before a crash and when crashing:
```
FW crashed - got error in interrupt endpoint thread function: status = 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED), actual = 0
```
```
13:43:49.050 [10981] [E] Device-5170: Error syncing TM2 and Host clocks
13:43:49.554 [10981] [E] Device-5170: ERROR: Bulk transfer message 0x2 (DEV_GET_TIME) request to device got LIBUSB_ERROR_IO. Bytes requested 6, bytes transferred 0
```
```
10:32:44.366 [20896] [E] Device-3FC0: FW crashed - got error in interrupt endpoint thread function: status = 0 (LIBUSB_SUCCESS / LIBUSB_TRANSFER_COMPLETED), actual = 0
10:32:44.596 [18550] [E] Device-3FC0: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
10:32:44.596 [18550] [E] Device-3FC0: Entered state [ERROR_STATE]
```

closed 2019-07-23 by joekeo

SteveGrehl, 2019-10-07:
@joekeo did you find a solution to this problem, since you closed the issue or is this a duplicate to #866 ?

yputra, 2019-11-04:
I also have the problem with Error:Bulk Transfer Message notifications after running around 5 mins, sometimes it can run up to 30 minutes, sometimes 10 minutes, it seems quite random, and happen after some accumulation of data ?

freemanlo, 2019-11-06:
@yputra,
1. Is your T265 connected to a USB3 hub? If so, is any other USB device connected?
2. What's your system configuration? OS? platform?
3. Which librealsense version do you use?
4. Do you run any librealsense app like realsense-viewer, rs-pose, etc to have this issue?

yputra, 2019-11-06:
Hi @freemanlo,

1. It is connected directly to USB3 port on the Raspberry Pi4, and there is another D435 camera connected there. I only use low res on D435 (420x240) and reduce the fps as well to minimize the data transfer.
2. it is on Raspberry Pi4b, with Ubuntu 18.04 and using ROS
3. It is using librealsense 2..29, I notice there is an update to 2.30 just yesterday, I will try that out tomorrow.
4. I only use the ros-package (realsense2_camera) and launch it to get the T265 and D435 launched. no other app.

freemanlo, 2019-11-08:
@yputra 

If you try the latest librealsense and still face the issue, you can also provide some debug logs.
by the way, what's your launch file command?

FPSychotic, 2020-02-14:
> Hi @freemanlo,
> 
> 1. It is connected directly to USB3 port on the Raspberry Pi4, and there is another D435 camera connected on another USB3 port. I only use low resolution on D435 (420x240) and reduce the fps as well to minimize the data transfer.
> 2. it is on Raspberry Pi4b, with Ubuntu 18.04 and using ROS
> 3. It is using librealsense 2.29.0, I notice there is an update to 2.30.0 yesterday, I will try that out tomorrow.
> 4. I only use the ros-package (realsense2_camera) and launch it to get the T265 and D435 launched. no other app.

did you solve this axactly same setup, same errors. I have a USB3.0 external powered, doesnt matter what I connect or disconect or were I connect them, it will crash

yputra, 2020-02-18:
@FPSychotic No, I still have the issue, and I have not got time to go back to it yet. I only tried on 2.32 once and it still crashed, so I put it on hold...

hmoyen, 2023-10-20:
any solutions for the problem? i am having random crashes using similar setup.

FPSychotic, 2023-10-20:
If you usa Raspberry PI and d435 +435  there is issues with RPI4 and USB physical address, specially when use USB poweres hub, if that is your setup, you cannot use RPI4