realsenseai/realsense-ros #826: How to use disparity filter twice to post process the depth image?

zinan93, 2019-06-24:
Hi, I know that filters can be specified in the launch file and that the parameters such as decimation scale can be set using dynamic reconfigure. But how do I convert the depth frame to disparity domain, do some temporal filtering, convert it back and then apply hole_filling? I tried the the following change in the rs_camera.launch file:

>   <arg name="filters"               default="decimation,disparity,spatial,temporal,disparity,hole_filling"/>

and got the following result:
[image]

it seems like the order of the sequence is not right, and only one disparity filter appeared.

doronhi, 2019-06-25:
Current filters interface is not as flexible as using the raw librealsense2 functionality.
Setting filter "disparity" encapsulate the other filters inside a disparity section, no matter how many times you specify disparity, it only appears once.
Decimation filter will be added only once and at the beginning of the filters queue.
You can check the code of the function [setupFilters](https://github.com/IntelRealSense/realsense-ros/blob/de1649a04ff7f96b549773fc709b3f8b2a81d397/realsense2_camera/src/base_realsense_node.cpp#L874) for a better understanding of the current situation and its limitations.

zinan93, 2019-06-26:
Ok I see, thank you. So the disparity filter is added to the beginning and end of the _filters vector after other filters has been pushed, then the decimation goes to the very beginning. Meaning the hole_filling filter will be applied in the disparity domain then.

closed 2019-06-26 by zinan93