realsenseai/realsense-ros #827: Can image resolution be changed dynamically?

xhuan28, 2019-06-25:
I would like to change the image resolution dynamically without stopping the realsense node. Is it possible to do that?

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Hi @ xhuan28
The resolution cannot be changed dynamically. Stop streaming is required by USB UVC protocol.

doronhi, 2019-10-30:
While, as said, you cannot change the output resolution of the device while streaming, you can use the decimation filter to change the resolution transmitted from realsense2_camera.

launch the driver with the decimation filter:
```roslaunch realsense2_camera rs_camera.launch filters:=decimation```

Then try out the effect of different parameters using rqt_reconfigure.
For color image change "stream_filter = Process - Color (2)", "stream_format_filter = Process - RGB8 (5)" and then play around with "filter_magnitude".

For depth resolution set "stream_filter = Process - Depth (1)", "stream_format_filter = Process - Z16 (1)"

RealSenseCustomerSupport, 2019-12-04:
----------------------------------------------

Hi @ xhuan28
Do you need further help on this?

xhuan28, 2019-12-05:
No, thanks. I think this ticket can be closed.

closed 2019-12-19 by RealSenseCustomerSupport