realsenseai/realsense-ros #828: T265 pose drift

AnthonyKyung11, 2019-06-25:
Hi, I'm currently working with the T265 module and I'm using it with SAMSUNG ARTIK 530.
The problem is when I run the roslaunch realsense2_camera rs_t265.launch the pose drifts significantly (I saw this issue #793 and I thought that would fix the problem, but it doesn't), but when I run the binary rs-pose the pose does not drifts.
I have the realsense-ros latest release (build 2.2.6), and have installed the ros-kinetic-ddynamic-reconfigure package.
Do you have any idea what could be causing this?

doronhi, 2019-06-26:
realsense2_camera does not affect T265 performance in terms of drifting and accuracy.
I assume the scenario you refer to is when the device stands still, without too much movement in it's view. This is when you see it's pose changes.
Since you mention that with rs-pose the pose does not drift, the main concern is that the librealsense2 version compiled with realsense2_camera is not the same as the one in rs-pose.
The log of realsense2_camera contains the version of librealsense2. Is it the same as the version shown in realsense-viewer?

AnthonyKyung11, 2019-07-02:
Thank you very much, that was it, I updated librealsense2 and it worked

closed 2019-07-02 by AnthonyKyung11