realsenseai/realsense-ros #829: Two T265 Cameras Publishing to ROS at the same time

asaba96, 2019-06-25:
----------------------------------------------------------------------------------------------------
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5715 | 
| Operating System |  Linux Ubuntu 18 | 
| ROS Version |  ROS Melodic | 
| Kernel Version (Linux Only)    |  4.18 | 
| Platform                                 | Intel NUC  |

### Issue Description
Hello, I have 2 tracking cameras connected to my Intel Nuc BEHi7. When I run `rs_mutlicam` example included in librealsense, I can stream pose and video feeds just fine from both cameras. 

However, when I launch `rs_mutliple_devices.launch`, only one set of camera topics shows up (i.e. `/camera1/odom/sample`) but not the other. I also tried specifying the serial numbers, launching each camera separately using `rs_t265.launch`, using the auto-discovery, etc. but nothing works. 

I installed librealsense via apt and built realsense_ros using the provided instructions. 

Interestingly, I keep getting the following error: 
`12:46:16.976 [14624] [E] Device-66E0: Error: Failed to claim USB interface. LIBUSB_ERROR_CODE: 0xFFFFFFFA (LIBUSB_ERROR_BUSY)`

Thank you for your help!

doronhi, 2019-06-26:
I'm sorry but it's not possible at the moment.
See #706

asaba96, 2019-06-28:
@doronhi Thank you, I must have missed that issue. Is there any way we can add this to the Readme? Correct me if I am wrong, but I read the Readme a few times and do not recall it being mentioned that this is a current issue with the T265 in particular.

doronhi, 2019-06-30:
You are correct, it's not in the README.md file and should rightly be there.
I hope the main problem will be fixed soon enough. Already there is a PR dealing with that issue - [#4302](https://github.com/IntelRealSense/librealsense/pull/4302)

asaba96, 2019-07-01:
@doronhi I opened #834 to add this to the README and I agree, I hope it gets addressed soon too. I have been enjoying using the cameras and would love to use multiple on some of our vehicles.

closed 2019-07-05 by asaba96