realsenseai/realsense-ros #830: D430 (NO RGB) overlay texturesource of infrared camera onto pointcloud/depthcloud/any depth source via MAVROS

kornee-mainblades, 2019-06-26:
Hi All,
I'm using a setup with a intelD4 board, and an intel D430 camera. Using the realsense-viewer software from Intel I can overlay the depth source in 3D with the infrared camera image.
However using the MAVROS software I don't see how to do this. Using the RGB(D) packages no relevant output is given since no RGD sensor is present in the camera.

Did any of you already use the D430 to produce an overlay of the infrared image onto a depthsouce/pointcloud?

Thanks in advance.

FPSychotic, 2019-07-17:
MAVROS I do not think it has something in common with rgbd packages or pointclouds or image topics.
 MAVROS just is for commands from the planners.

kornee-mainblades, 2019-07-18:
sorry I was referring to the ROS wrapper

doronhi, 2019-07-18:
running the following command:
```
roslaunch realsense2_camera rs_camera.launch filters:=pointcloud pointcloud_texture_stream:=RS2_STREAM_INFRARED
```
Will show the pointcloud with infrared image as texture.

FPSychotic, 2019-07-18:
Wow, it means that I release the RGB camera from the depth processing and use it foe example for CV, video recording etc?
I thought in open a question for this.
I want that just stereo cameras for mapping and use the RGB for object recognition via OpenCV or CNN

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Hi @FPSychotic,

Is the issue clarified/resolved per @doronhi's suggestion? To apply IR texture to depth could have the IR projector's pattern impact.
If no further question, will close this.

closed 2019-12-05 by RealSenseCustomerSupport